hanyazou/TelloPy #1: Can't Install

lpmrz, 2018-05-29:
Hello,

When I type, 'pip install tellopy', I get this;

Collecting tellopy
  Could not find a version that satisfies the requirement tellopy (from versions: )
No matching distribution found for tellopy

Can you help please?

Thank you

hanyazou, 2018-06-01:
Tellopy supports python2 for now. Tellopy depends on louie which can't run on python 3.

>$ python3 -m pip install tellopy
Collecting tellopy
  Could not find a version that satisfies the requirement tellopy (from versions: )
No matching distribution found for tellopy

>$ python2 -m pip install tellopy
Collecting tellopy
  Downloading https://files.pythonhosted.org/packages/44/22/017cccb054f710ca254327aef41b03998edf5068803c1c6193ea2915d0f4/tellopy-0.3.0-py2.py3-none-any.whl
Requirement already satisfied: louie in /usr/local/lib/python2.7/site-packages (from tellopy) (1.1)
Requirement already satisfied: nose>=0.8.3 in /usr/local/lib/python2.7/site-packages (from louie->tellopy) (1.3.7)
Installing collected packages: tellopy
Successfully installed tellopy-0.3.0

hanyazou, 2018-06-01:
Louie has been modified to support python 3 in 2016. But that does not uploaded to PyPi yet...
 
https://github.com/11craft/louie/issues/6

hanyazou, 2018-06-02:
New version 0.4.0 was released, which supports python 3.6.

closed 2018-06-02 by hanyazou