hanyazou/TelloPy #5: opencv av stream

vtushevskiy, 2018-06-13:
Hi!
Thanks for amazing SDK!
There is an issue in video_effect.py: the video by default has very big lag. It tries to work on every single frame with huge resolution and it has certain delay.
I guess we could use something like this but more sophisticated like adaptive frame skip:
`        while True:
            i = 0
            for frame in container.decode(video=0):
                print i
                i=i+1
                if i>300: #skip first 300 frames
                    if i%4==0: #do only 1/4 frames
                        im = numpy.array(frame.to_image())
                        im = cv2.resize(im, (320,240)) #resize frame
                        image = cv2.cvtColor(im, cv2.COLOR_RGB2BGR)
                        cv2.imshow('Original', image)
                        cv2.imshow('Canny', cv2.Canny(image, 100, 200))
                        cv2.waitKey(1)`

hanyazou, 2018-06-16:
Thank you for good suggestion.

I've been concerned about the delay. Skipping first frames seems to be enough on my Mac to fix the delay.

        i = 0
        while True:
            for frame in container.decode(video=0):
               # skip first 300 frames
                i = i + 1
                if i < 300:
                    continue
                image = cv2.cvtColor(numpy.array(frame.to_image()), cv2.COLOR_RGB2BGR)
                cv2.imshow('Original', image)
                cv2.imshow('Canny', cv2.Canny(image, 100, 200))
                cv2.waitKey(1)

Do you really need skip 3/4 frames and resize? 
My mac is iMac late 2012 3.4GHz Intel Core i7. Please tell me about your machine.

hanyazou, 2018-06-16:
I modified video_effect.py at 8dacdf9

vtushevskiy, 2018-06-19:
Actually it works for me with every frame

hanyazou, 2018-06-23:
so we can close this issue.

closed 2018-06-23 by hanyazou