hanyazou/TelloPy #6: mac os ps4 joystick works only in wired mode

vtushevskiy, 2018-06-14:
Mac os ps4 joystick works only in wired mode. When wireless it is detecting but no response to buttons/axises.

hanyazou, 2018-06-15:
I can use ps4 joystick on my Mac via bluetooth. I've tested it on Python 2.7 and 3.6 today. Did you try other application to test the joystick? I can see my joystick input with "Joystick Show"

https://itunes.apple.com/us/app/joystick-show/id515886877

edmondja, 2018-06-26:
I can relate, it only works fine in wired mode. 
[https://ucc65f8e446c5952cc095b7ac5fc.dl.dropboxusercontent.com/cd/0/get/AJ3ob-lTstWSIZF58EybWRKIpKeslHkii6fzo90O_syG6XBcjLB00Pwr7TSIO8H9JMubtXxLzr2RsHfv5kV79TPaKApzu46AABI97DSlz5rvwZyeiPzBOLVz90y42H31CeHhI8R-eoPyzW6140Zb0IPpfXM8cgFynq4Fhl11DNKsEzmqeGXiDAiPumg9tn1bOS4/file?_download_id=034284071934807656931779427537053188320690867981539405642054916157&_notify_domain=www.dropbox.com&dl=1](url)

hanyazou, 2018-06-27:
Sorry. I can't understand what you mean. And the URL you provide is 404 error.

edmondja, 2018-06-27:
Oh, I just wanted to confirm you what he said, and that even Joystick Show works properly in my macbook.

edmondja, 2018-06-27:
Same os X version.

hanyazou, 2018-06-30:
Same Mac OS version. It works on my iMac. What's the difference?

> bash-3.2$ python2 -m tellopy.examples.joystick_and_video
2018-06-30 14:26:29.673 Python[3528:340370] ApplePersistence=NO
Joystick name: Wireless Controller
bash-3.2$ python3 -m tellopy.examples.joystick_and_video
2018-06-30 14:25:44.964 Python[3505:339571] ApplePersistence=NO
Joystick name: Wireless Controller
bash-3.2$ pip2 install pygame
Requirement already satisfied: pygame in /usr/local/lib/python2.7/site-packages (1.9.3)
bash-3.2$ pip3 install pygame
Requirement already satisfied: pygame in /usr/local/lib/python3.6/site-packages (1.9.3)

[image]
[image]

@edmondja 
> the reactivity is much better when we get rid of the time.sleep(0.01) line 179 (I dont understand why its there..)

Without any sleep in the loop, the loop consume CPU time too much. In fact, Video decoding performance was not good before I put the sleep there.
I think 100Hz is enough in this case.  But you can reduce the sleep duration to 0.001 (1ms/1000Hz) if you need.

edmondja, 2018-06-30:
Ok thanks.
I use python 3 and I have the same pygame version. 
On bluetooth :  

$ python -m tellopy.examples.joystick_and_video
Joystick name: Wireless Controller
Tello: 09:27:51.861:  Info: start video thread
Tello: 09:27:51.861:  Info: send connection request (cmd="conn_req:9617")
Tello: 09:27:51.861:  Info: video receive buffer size = 524288
Tello: 09:27:51.861:  Info: state transit State::disconnected -> State::connecting
Tello: 09:27:51.861:  Info: start video (cmd=0x25 seq=0x01e4)
height= 0, fly_mode=0x01, battery_percentage= 4, drone_battery_left=0x0000
Tello: 09:27:51.874:  Info: connected. (port=9617)
Tello: 09:27:51.875:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 09:27:51.875:  Info: state transit State::connecting -> State::connected
Tello: 09:27:51.875:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 09:27:51.876:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 09:27:51.926:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 09:27:51.927:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
MPlayer 1.3.0-4.2.1 (C) 2000-2016 MPlayer Team
Can't init Apple Remote.

Playing -.
Reading from stdin...
libavformat version 57.25.100 (internal)
Cannot seek backward in linear streams!
Seek failed
Cannot seek backward in linear streams!
Seek failed

How can I debug it in less lines of code to isolate the problem ?

vtushevskiy, 2018-06-30:
As I understand, upgrading from python 2.7 to python 3.6 fixed problem with joystick in wireless mode?

hanyazou, 2018-06-30:
Python 2.7 and pygame 1.9.3 works fine with PS4 controller via BT in my environment.

>bash-3.2$ python2 -m tellopy.examples.joystick_and_video
2018-06-30 21:32:19.752 Python[956:9546] ApplePersistence=NO
Joystick name: Wireless Controller
Tello: 21:32:20.156:  Info: start video thread
Tello: 21:32:20.156:  Info: send connection request (cmd="conn_req:9617")
Tello: 21:32:20.156:  Info: video receive buffer size = 524288
Tello: 21:32:20.156:  Info: state transit State::disconnected -> State::connecting
Tello: 21:32:20.156:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 21:32:20.328:  Info: connected. (port=9617)
Tello: 21:32:20.328:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 21:32:20.328:  Info: state transit State::connecting -> State::connected
Tello: 21:32:20.329:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 21:32:20.329:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
height= 0, fly_mode=0x01, battery_percentage=83, drone_battery_left=0x0000
Tello: 21:32:20.353:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 21:32:20.354:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
height= 1, fly_mode=0x01, battery_percentage=83, drone_battery_left=0x0000
height= 2, fly_mode=0x01, battery_percentage=83, drone_battery_left=0x0000
height= 0, fly_mode=0x01, battery_percentage=83, drone_battery_left=0x0000
Tello: 21:32:21.325:  Info: video recv: 1460 bytes f600 +151
MPlayer 1.3.0-4.2.1 (C) 2000-2016 MPlayer Team
Can't init Apple Remote.
>
>Playing -.
Reading from stdin...
libavformat version 57.25.100 (internal)
height=65535, fly_mode=0x01, battery_percentage=83, drone_battery_left=0x0000
Cannot seek backward in linear streams!
Seek failed

hanyazou, 2018-08-09:
Please re-open when you have any updates.

closed 2018-08-09 by hanyazou

akashliev, 2019-08-29:
I have the same issue. PS4 wireless dual shock controller. This worked:
python -m tellopy.examples.simple_takeoff

But this doesn't:
python -m tellopy.examples.joystick_and_video

I paired the controller with my MacBook Pro, ran the above command, the code runs but when I press any button on my PS4 controller nothing happens.

hanyazou, 2019-08-31:
Hi @akashliev 
You should upload all output of the execution.
Did you try any other joy stick application or tester on the Mac?