hanyazou/TelloPy #8: Regular image sampling and latency

edmondja, 2018-06-17:
Is there a way to slow down the for loop in the video streaming example : 
"for frame in container.decode(video=0):" so we can use heavy slow computations in each iteration of the loop and keep having a regularly sampled images stream ?

Or maybe we could use http://www.scikit-video.org/stable/ for this instead ?

By the way, do we have an idea of the latency between each frame being downloaded from the drone to the computer ?

hanyazou, 2018-06-23:
How about adding a thread? This works for me.

https://github.com/hanyazou/TelloPy/blob/8f0ae04552972143baadcafd73238a953ba6c48b/tellopy/examples/video_effect.py

edmondja, 2018-06-23:
Wow thanks, I’ll try to see if this works :)

hanyazou, 2018-06-23:
Sorry. Line 57 `image = None` in that source is wrong. It should be `frame = None`.

edmondja, 2018-06-25:
Hey, I took a look at you solution, I think yours is more elegant than what I did before but I don't really see a difference.

Just to understand, why do you use threading instead of just skipping the images which didn't appear until the timer is passed ? like this : 

starting = time.time()
for frame in container.decode(video=0):
    timer = time.time()

    if (timer - starting) * 1000 > 100 :
        starting = timer
        # Here do some heavy computations (heavy but still <100ms)

vtushevskiy, 2018-06-25:
Thanks for threading example!!

vtushevskiy, 2018-06-25:
I just checked threading example. 
I did not test it jet, but will there be no conflict of concurrent access to 'frame' variable? 
Shouldn't you lock it when write/read?

vtushevskiy, 2018-06-25:
Also, I believe there is no necessarity to skip first 300 frames when we use threading. OpenCV part will just access to last available frame.

vtushevskiy, 2018-06-26:
Just tested it: works perfect. I don't know why but there are no concurrent access conflicts. Also, skipping of first 300 frames is not necessary anymore, there is no lag.
Great improvement!

hanyazou, 2018-06-30:
I've modified the video_effect.py example on develop-0.5.0 branch.

https://github.com/hanyazou/TelloPy/commit/4f0609e74ddc0beeac9100cb2c04c514754b6f3b

The idea is that it skips subsequent frames if it consume processing time longer than the frame interval. This idea is based on the post by @edmondja.

Thank you, @edmondja !

edmondja, 2018-06-30:
By the way, does the fact of using the thread  https://github.com/hanyazou/TelloPy/issues/8#issuecomment-399617341 consumes too much CPU the same way the while loop here https://github.com/hanyazou/TelloPy/issues/6#issuecomment-401519284 when you get rid of the sleep ?

hanyazou, 2018-06-30:
@edmondja 
I think every thread should have some sleep or yield in it's loop. I guess that video decoder have some sleep in it. It is waiting for some input from the source.

closed 2018-06-30 by edmondja