hanyazou/TelloPy #9: downgrade python or anything

eggs23, 2018-06-21:
after wrote the line ; $ pip install dist/tellopy-0.5.0.dev*.whl --upgrade
 i got the error
tellopy requires Python '>=2.7', !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*, <4' but the running is 3.5.3

How can i solve this problem ?

hanyazou, 2018-06-22:
There is a line describing required python version in the setup.py.

> python_requires='>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*, <4',

This line prevent you from installing the TelloPy on to Python 3.5. I made a new branch develop-0.5.0-python3_5, in which branch I just delete "!=3.5.*," to allow installing. I don't know if the TelloPy run on Python 3.5.

You can try the branch like bellow. Please test this and report if you can use TelloPy on python 3.5 so that I can update and release TelloPy for python 3.5.

> bash-3.2$ **git clone -b develop-0.5.0-python3_5 https://github.com/hanyazou/TelloPy** 
Cloning into 'TelloPy'...
remote: Counting objects: 213, done.        
remote: Compressing objects: 100% (123/123), done.        
remote: Total 213 (delta 118), reused 179 (delta 86), pack-reused 0        
Receiving objects: 100% (213/213), 3.95 MiB | 3.01 MiB/s, done.
Resolving deltas: 100% (118/118), done.
bash-3.2$ **pip install -e TelloPy/ --upgrade**
Obtaining file:///private/tmp/TelloPy
Installing collected packages: tellopy
  Found existing installation: tellopy 0.5.0.dev0
    Uninstalling tellopy-0.5.0.dev0:
      Successfully uninstalled tellopy-0.5.0.dev0
  Running setup.py develop for tellopy
Successfully installed tellopy
bash-3.2$ **python -m tellopy.examples.simple_takeoff**
Tello: 08:20:16.395:  Info: start video thread
Tello: 08:20:16.395:  Info: send connection request (cmd="conn_req:9617")
Tello: 08:20:16.395:  Info: video receive buffer size = 524288
Tello: 08:20:16.395:  Info: state transit State::disconnected -> State::connecting
Tello: 08:20:18.395:  Info: send connection request (cmd="conn_req:9617")
Tello: 08:20:20.395:  Info: send connection request (cmd="conn_req:9617")

eggs23, 2018-06-23:
when i try the second old error occured:

import av
ImporError: no module named 'av'

hanyazou, 2018-06-23:
Could you provide more specific command line which you try and output messages?

eggs23, 2018-07-01:
$python3 -m tellopy.examples.video_effect

Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.5/dist-packages/tellopy/examples/video_effect.py", line 5, in <module>
    import cv2.cv2 as cv2  # for avoidance of pylint error
  File "/usr/local/lib/python3.5/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: libQtGui.so.4: cannot open shared object file: No such file or directory

hanyazou, 2018-07-01:
So the tellopy work on Python 3.5 in your environment? I will merge the develop-0.5.0-python3_5 branch after your confirmation.
And the latency problem was fixed on newer version of develop-0.5.0. (see #8)

eggs23, 2018-07-02:
Yes it works on Python 3.5.
Thank you

hanyazou, 2018-07-04:
branch develop-0.5.0 was updated in 06acbade59ab80aeb0c57a84b333a1b77e91955d

closed 2018-07-04 by hanyazou