hanyazou/TelloPy #10: Weird event datas

edmondja, 2018-07-01:
I wanted to gather more data about flights but it seems that they are weirdly encoded when I follow the "joystick" example.

```
def handler(event, sender, data, **args):
    global prev_flight_data
    global video_player
    global prev_ev_connected
    global prev_EVENT_LIGHT
    global prev_EVENT_LOG
    global prev_EVENT_TIME
    global prev_FLIGHT_EVENT
    global prev_EVENT_WIFI
    drone = sender
    if event is drone.EVENT_FLIGHT_DATA:
        if prev_flight_data != str(data):
            print('les flight valent')
            print(data)
            prev_flight_data = str(data)
    elif event is drone.EVENT_LIGHT:
        if prev_EVENT_LIGHT != str(data):
            print('les EVENT_LIGHT datas valent')
            print(data)
            prev_EVENT_LIGHT = str(data)
    elif event is drone.EVENT_LOG:
        if prev_EVENT_LOG != str(data):
            print('les prev_EVENT_LOG datas valent')
            print(data)
            prev_EVENT_LOG = str(data)
```

I receive strings of this format : b",C\x00'\x00\x00\x00\x03\x00\x00\x00*\x00\x00\x00\x00\x00\x00\x00BUILD Apr 23 2018 12:19:39\x00\x00\x00\x00\x00\x00F\xc3#\x00\x07\x00\x00\x00\x00\x03e\x01\x1b\x16\x13`\x00eebf\x1b\x11\x00UU\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaaUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\x90\x07\x00\x00\x8f\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00DJI_LOG_V3\x92\xdbK\t\xfd\xe9" .

"height= 2, fly_mode=0x0c, battery_percentage=74, drone_battery_left=0x0000" <- even flight datas actually aren't properly formatted.

What can I do to fix it ? Wifi strength is for example something I really needed.

hanyazou, 2018-07-01:
Please refer definition of FlightData class in tellopy/_internal/protocol.py. String representation of the class instance is given by __str__() method. You can add more members in it's __str__() method or you could just delete the method. But wifi_strength member is not filled from flight data message. I don't have any idea about the members in the class because it was just brought from original Golang version.

edmondja, 2018-07-01:
Thanks, I tried and indeed wifi_strength is always 0.

closed 2018-07-01 by edmondja