hanyazou/TelloPy #12: Error installing module av

douglasrubims, 2018-07-11:
`Microsoft Windows [versão 10.0.17134.165]
(c) 2018 Microsoft Corporation. Todos os direitos reservados.

C:\Users\dougl>pip install av
Collecting av
  Using cached https://files.pythonhosted.org/packages/52/56/3ed0d68acc9d1faa92b15e7012c68fad5f21c0405315c90b337104ba7f29/av-0.4.1.tar.gz
Installing collected packages: av
  Running setup.py install for av ... error
    Complete output from command c:\python27\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\dougl\\appdata\\local\\temp\\pip-install-mbxiy2\\av\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\dougl\appdata\local\temp\pip-record-unow7x\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-2.7
    creating build\lib.win-amd64-2.7\av
    copying av\__init__.py -> build\lib.win-amd64-2.7\av
    copying av\__main__.py -> build\lib.win-amd64-2.7\av
    creating build\lib.win-amd64-2.7\av\audio
    copying av\audio\__init__.py -> build\lib.win-amd64-2.7\av\audio
    creating build\lib.win-amd64-2.7\av\codec
    copying av\codec\__init__.py -> build\lib.win-amd64-2.7\av\codec
    creating build\lib.win-amd64-2.7\av\container
    copying av\container\__init__.py -> build\lib.win-amd64-2.7\av\container
    creating build\lib.win-amd64-2.7\av\data
    copying av\data\__init__.py -> build\lib.win-amd64-2.7\av\data
    creating build\lib.win-amd64-2.7\av\filter
    copying av\filter\__init__.py -> build\lib.win-amd64-2.7\av\filter
    creating build\lib.win-amd64-2.7\av\subtitles
    copying av\subtitles\__init__.py -> build\lib.win-amd64-2.7\av\subtitles
    creating build\lib.win-amd64-2.7\av\video
    copying av\video\__init__.py -> build\lib.win-amd64-2.7\av\video
    running build_ext
    running reflect
    running config
    looking for avformat_open_input... missing
    looking for pyav_function_should_not_exist... missing
    looking for av_calloc... missing
    looking for av_frame_get_best_effort_timestamp... missing
    looking for avformat_alloc_output_context2... missing
    looking for avformat_close_input... missing
    looking for avcodec_send_packet... missing
    looking for AV_OPT_TYPE_INT...missing
    looking for PYAV_ENUM_SHOULD_NOT_EXIST...missing
    looking for AV_OPT_TYPE_BOOL...missing
    looking for AVStream.index... missing
    looking for PyAV.struct_should_not_exist... missing
    looking for AVFrame.mb_type... missing

    We didn't find `avformat_open_input` in the libraries.
    We look for it only as a sanity check to make sure the build
    process is working as expected. It is not, so we must abort.

    You can see the compiler output for the reflection process via:
        python setup.py reflect --force --debug

    Here is the config we gathered so far:

    PyAV: 0.4.1 (unknown commit)
    Python: 2.7.15 (v2.7.15:ca079a3ea3, Apr 30 2018, 16:30:26) [MSC v.1500 64 bit (AMD64)]
    platform: Windows-10-10.0.17134
    extension_extra:
        extra_link_args: ['/OPT:NOREF']
        define_macros: []
        runtime_library_dirs: []
        libraries: ['avformat', 'avcodec', 'avdevice', 'avutil', 'avfilter', 'swscale', 'swresample']
        library_dirs: ['c:\\python27\\libs', 'c:\\python27\\PCbuild\\amd64', 'c:\\python27\\PC\\VS9.0\\amd64']
        include_dirs: ['include', 'c:\\python27\\include', 'c:\\python27\\PC']
    config_macros:
        PYAV_COMMIT_STR="unknown-commit"
        PYAV_HAVE_AVCODEC_SEND_PACKET=0
        PYAV_HAVE_AVFORMAT_ALLOC_OUTPUT_CONTEXT2=0
        PYAV_HAVE_AVFORMAT_CLOSE_INPUT=0
        PYAV_HAVE_AVFRAME__MB_TYPE=0
        PYAV_HAVE_AV_CALLOC=0
        PYAV_HAVE_AV_FRAME_GET_BEST_EFFORT_TIMESTAMP=0
        PYAV_HAVE_AV_OPT_TYPE_BOOL=0
        PYAV_HAVE_LIBAVRESAMPLE=0
        PYAV_HAVE_LIBSWRESAMPLE=1
        PYAV_VERSION=0.4.1
        PYAV_VERSION_STR="0.4.1"
        inline=__inline

    ----------------------------------------
Command "c:\python27\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\dougl\\appdata\\local\\temp\\pip-install-mbxiy2\\av\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record c:\users\dougl\appdata\local\temp\pip-record-unow7x\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in c:\users\dougl\appdata\local\temp\pip-install-mbxiy2\av\

C:\Users\dougl>`

Could you please help me?

hanyazou, 2018-07-11:
I just googled and found a similar issue on the official PyAv github page.

https://github.com/mikeboers/PyAV/issues/199

Does this help you?

douglasrubims, 2018-07-11:
I'd already tried that.
I was able to install Pyav with ffmpeg (because I can not install Conda), but I have this problem:

`C:\Users\dougl\Downloads\PyAV-develop>python
Python 2.7.15 (v2.7.15:ca079a3ea3, Apr 30 2018, 16:30:26) [MSC v.1500 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import av
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "av\__init__.py", line 9, in <module>
    from av._core import time_base, pyav_version as __version__
ImportError: No module named _core
>>>`

I could not find a solution to this ...
I'm sorry if my English is bad, because I do not speak this language, I'm Brazilian.

douglasrubims, 2018-07-12:
Hello, finally I was able to install the AV module, but for some reason I can not access it :( could you help me with this?

[image]

GalBrandwine, 2018-07-21:
Hey there, 
first, grate work on tellopy!
second, I had the same problem. after pip-installing  av, and error: 
(working on ubuntu 16.x)

> pip install av
> Collecting av
>   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x7fa1fa71e518>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)': /simple/av/
>   Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x7fa1fa71e8d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)': /simple/av/
>   Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x7fa1fa71e978>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)': /simple/av/
>   Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x7fa1fa71ed68>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)': /simple/av/
>   Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x7fa1fa71eef0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)': /simple/av/
>   Could not find a version that satisfies the requirement av (from versions: )
> No matching distribution found for av
> (droneProj1) gal@Gal-pc:~$ pip install av
> Collecting av
>   Using cached https://files.pythonhosted.org/packages/52/56/3ed0d68acc9d1faa92b15e7012c68fad5f21c0405315c90b337104ba7f29/av-0.4.1.tar.gz
> Building wheels for collected packages: av
>   Running setup.py bdist_wheel for av ... error
>   Complete output from command /home/gal/.virtualenvs/droneProj1/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-aqq8h7iq/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-vnpf2n4q --python-tag cp35:
>   running bdist_wheel
>   running build
>   running build_py
>   creating build
>   creating build/lib.linux-x86_64-3.5
>   creating build/lib.linux-x86_64-3.5/av
>   copying av/__init__.py -> build/lib.linux-x86_64-3.5/av
>   copying av/__main__.py -> build/lib.linux-x86_64-3.5/av
>   creating build/lib.linux-x86_64-3.5/av/data
>   copying av/data/__init__.py -> build/lib.linux-x86_64-3.5/av/data
>   creating build/lib.linux-x86_64-3.5/av/container
>   copying av/container/__init__.py -> build/lib.linux-x86_64-3.5/av/container
>   creating build/lib.linux-x86_64-3.5/av/filter
>   copying av/filter/__init__.py -> build/lib.linux-x86_64-3.5/av/filter
>   creating build/lib.linux-x86_64-3.5/av/video
>   copying av/video/__init__.py -> build/lib.linux-x86_64-3.5/av/video
>   creating build/lib.linux-x86_64-3.5/av/audio
>   copying av/audio/__init__.py -> build/lib.linux-x86_64-3.5/av/audio
>   creating build/lib.linux-x86_64-3.5/av/subtitles
>   copying av/subtitles/__init__.py -> build/lib.linux-x86_64-3.5/av/subtitles
>   creating build/lib.linux-x86_64-3.5/av/codec
>   copying av/codec/__init__.py -> build/lib.linux-x86_64-3.5/av/codec
>   running build_ext
>   running reflect
>   running config
>   Could not find libavdevice with pkg-config.
>   Could not find libavfilter with pkg-config.
>   
>   ----------------------------------------
>   Failed building wheel for av
>   Running setup.py clean for av
> Failed to build av
> Installing collected packages: av
>   Running setup.py install for av ... error
>     Complete output from command /home/gal/.virtualenvs/droneProj1/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-aqq8h7iq/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-zyy44axt/install-record.txt --single-version-externally-managed --compile --install-headers /home/gal/.virtualenvs/droneProj1/include/site/python3.5/av:
>     running install
>     running build
>     running build_py
>     creating build
>     creating build/lib.linux-x86_64-3.5
>     creating build/lib.linux-x86_64-3.5/av
>     copying av/__init__.py -> build/lib.linux-x86_64-3.5/av
>     copying av/__main__.py -> build/lib.linux-x86_64-3.5/av
>     creating build/lib.linux-x86_64-3.5/av/data
>     copying av/data/__init__.py -> build/lib.linux-x86_64-3.5/av/data
>     creating build/lib.linux-x86_64-3.5/av/container
>     copying av/container/__init__.py -> build/lib.linux-x86_64-3.5/av/container
>     creating build/lib.linux-x86_64-3.5/av/filter
>     copying av/filter/__init__.py -> build/lib.linux-x86_64-3.5/av/filter
>     creating build/lib.linux-x86_64-3.5/av/video
[... 2 lines cut ...]
>     copying av/audio/__init__.py -> build/lib.linux-x86_64-3.5/av/audio
>     creating build/lib.linux-x86_64-3.5/av/subtitles
[... cut ...] naged --compile --install-headers /home/gal/.virtualenvs/droneProj1/include/site/python3.5/av" failed with error code 1 in /tmp/pip-install-aqq8h7iq/av/

using [this site](https://mikeboers.github.io/PyAV/installation.html#ubuntu-14-04-lts), ive fixed the problem by:

> sudo apt-get install -y python-dev pkg-config
> 
> sudo apt-get install -y \
>     libavformat-dev libavcodec-dev libavdevice-dev \
>     libavutil-dev libswscale-dev libavresample-dev libavfilter-dev

and than:

> pip install av

hanyazou, 2018-09-29:
Close this issue because @GalBrandwine show the solution.

closed 2018-09-29 by hanyazou

nehaanand1719, 2020-02-15:
> I'd already tried that.
> I was able to install Pyav with ffmpeg (because I can not install Conda), but I have this problem:
> 
> `C:\Users\dougl\Downloads\PyAV-develop>python
> Python 2.7.15 (v2.7.15:ca079a3ea3, Apr 30 2018, 16:30:26) [MSC v.1500 64 bit (AMD64)] on win32
> Type "help", "copyright", "credits" or "license" for more information.
> 
> > > > import av
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "av__init__.py", line 9, in 
> > > > from av._core import time_base, pyav_version as **version**
> > > > ImportError: No module named _core
> > > > `
> 
> I could not find a solution to this ...
> I'm sorry if my English is bad, because I do not speak this language, I'm Brazilian.

How have u installed ?

douglasrubims, 2020-02-18:
Portuguese:
No windows, o que você pode fazer é baixar o FFMPEG e colocar seu conteúdo em C:\ffmpeg, então você coloca "C:\ffmpeg\bin" na PATH do windows e instala o Anaconda (conda), e através dele você pode instalar o módulo PyAV.
Qualquer dúvida, me envie por aqui.

English:
On windows, what you can do is download FFMPEG and put its content in "C:\ffmpeg", then you put "C:\ffmpeg\bin" in the Windows PATH and install Anaconda (conda), and through it you can install the PyAV module.
Any questions, send me here.

douglasrubims, 2020-02-18:
> How have u installed ?

Portuguese:
No windows, o que você pode fazer é baixar o FFMPEG e colocar seu conteúdo em C:\ffmpeg, então você coloca "C:\ffmpeg\bin" na PATH do windows e instala o Anaconda (conda), e através dele você pode instalar o módulo PyAV.
Qualquer dúvida, me envie por aqui.

English:
On windows, what you can do is download FFMPEG and put its content in "C:\ffmpeg", then you put "C:\ffmpeg\bin" in the Windows PATH and install Anaconda (conda), and through it you can install the PyAV module.
Any questions, send me here.

nehaanand1719, 2020-02-18:
Any other wqy without Conda ?

douglasrubims, 2020-02-18:
> Any other wqy without Conda ?
> […](#)

It is possible if you install the Windows SDK (which includes the Windows compiler kit) and MinGW .. but my tip is: compile PyAV manually.

nehaanand1719, 2020-02-18:
What did u mwan by compiling it manually ?

douglasrubims, 2020-02-18:
> What did u mwan by compiling it manually ?
> […](#)

In end of this page has a tutorial:
https://docs.mikeboers.com/pyav/develop/overview/installation.html

nehaanand1719, 2020-02-18:
Tried this way,but getting error .
Cannot open file .lib

MoinShah031, 2020-03-01:
> Portuguese:
> No windows, o que você pode fazer é baixar o FFMPEG e colocar seu conteúdo em C:\ffmpeg, então você coloca "C:\ffmpeg\bin" na PATH do windows e instala o Anaconda (conda), e através dele [...]

please explain How to install Av package to python 3.8. its shows error

douglasrubims, 2020-03-16:
I went to the Microsoft Store and downloaded Python 3.8, right after I used the command "python3 -m pip install av" and it was downloaded and installed without any problems. My theory is that I already have all the necessary compilers on my system.
Compilers can be downloaded from this link:
https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=BuildTools&rel=16

Image showing my installation:
[image]

I apologize for the delay in responding, I have been very busy with college for the past few days.
Hope this helps!

MoinShah031, 2020-03-16:
Thanks sir, but sir I am working from last 10 days. I install openvino which is only compatible with python version 3.6Then different kind of error occur such is import openVino, IE network and Liv. Camera etcSent from my Huawei phone-------- Original message --------From: Douglas Rubim <notifications@github.com>Date: Mon, Mar 16, 2020, 5:24 AMTo: hanyazou/TelloPy <TelloPy@noreply.github.com>Cc: MoinShah031 <moinshah031@gmail.com>, Comment <comment@noreply.github.com>Subject: Re: [hanyazou/TelloPy] Error installing module av (#12)
I went to the Microsoft Store and downloaded Python 3.8, right after I used the command "python3 -m pip install av" and it was downloaded and installed without any problems. My theory is that I already have all the necessary compilers on my system.
Compilers can be downloaded from this link:
https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=BuildTools&rel=16
Image showing my installation:

I apologize for the delay in responding, I have been very busy with college for the past few days.
Hope this helps!

MoinShah031, 2020-03-16:
Sir I am student of electrical engineering and my programming skills is weak. Sir please if you have some leasure time I connect my computer screen with you with any disk for removing of error. Please help this is very meaningful for meSent from my Huawei phone-------- Original message --------From: Douglas Rubim <notifications@github.com>Date: Mon, Mar 16, 2020, 5:24 AMTo: hanyazou/TelloPy <TelloPy@noreply.github.com>Cc: MoinShah031 <moinshah031@gmail.com>, Comment <comment@noreply.github.com>Subject: Re: [hanyazou/TelloPy] Error installing module av (#12)
I went to the Microsoft Store and downloaded Python 3.8, right after I used the command "python3 -m pip install av" and it was downloaded and installed without any problems. My theory is that I already have all the necessary compilers on my system.
Compilers can be downloaded from this link:
https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=BuildTools&rel=16
Image showing my installation:

I apologize for the delay in responding, I have been very busy with college for the past few days.
Hope this helps!

amaximus, 2021-07-08:
Try:

`sudo apt-get install libavformat-dev libavdevice-dev libavcodec-dev libavfilter-dev libavutil-dev libswscale-dev libswresample-dev` then `pip install av`