hanyazou/TelloPy #13: Does the video stream function work with Wifi Adapter?

francisduan, 2018-07-19:
Hey really cool design. I have used the scripts well on my laptop, but when I transferred everything to my PC, which uses a Wifi adapter to communicate with Tello, the video stream function seems to be not working. The other commands such as take off and moving functions work fine, it is only the video stream function that I could not duplicate. Could you help me a bit on this?

Tello: 14:02:56.879:  Info: start video thread
Tello: 14:02:56.879:  Info: send connection request (cmd="conn_req:9617")
Tello: 14:02:56.880:  Info: video receive buffer size = 524288
Tello: 14:02:56.880:  Info: state transit State::disconnected -> State::connecting
Tello: 14:02:56.885:  Info: connected. (port=9617)
Tello: 14:02:56.885:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 14:02:56.885:  Info: state transit State::connecting -> State::connected
connected
Tello: 14:02:56.885:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 14:02:56.885:  Info: set exposure (cmd=0x34 seq=0x01e4)
Tello: 14:02:56.886:  Info: set video encoder rate (cmd=0x20 seq=01e4)
Tello: 14:02:56.889:  Info: get video stream
Tello: 14:02:56.889:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 14:02:56.936:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 14:02:56.937:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 14:02:56.938:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 14:02:56.939:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 14:02:56.939:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 14:02:56.940:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 14:02:56.941:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 14:02:56.941:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
event="time" data=52480
height= 0, fly_mode=0x01, battery_percentage=56, drone_battery_left=0x0000
event="time" data=59392
height= 0, fly_mode=0x01, battery_percentage=55, drone_battery_left=0x0000
event="time" data=1536
Tello: 14:03:02.882: Error: video recv: timeout
event="time" data=8448
event="time" data=15360
Tello: 14:03:07.882: Error: video recv: timeout
event="time" data=22272
event="time" data=29440
Tello: 14:03:12.883: Error: video recv: timeout

hanyazou, 2018-09-29:
I have no idea about your wifi trouble, sorry. But @ToxicFrog implemented the still image capture function. Please search 'tale_picture' in his sample app to find how to use the function.

https://github.com/hanyazou/TelloPy/blob/develop-0.6.0/tellopy/examples/keyboard_and_video.py

ToxicFrog, 2018-09-30:
Bear in mind that you will need to permit your firewall to allow outgoing UDP traffic to, _and incoming UDP traffic from_, the Tello; if it works on one machine but not on the other, that's the most likely cause, and the symptoms you describe sound like your firewall is letting the computer send commands to the Tello but rejecting the image packets the Tello is sending to the computer.

Try opening inbound UDP ports 9000, 9617, and 6038 -- or just temporarily disabling the firewall for the wifi adapter -- and see if that helps.

francisduan, 2018-09-30:
Thanks for the help! It worked!

Francis

closed 2018-09-30 by hanyazou

francisduan, 2018-10-09:
Hey Ben,

I am trying to to feed the live stream into Unity. So that users could see
what tells sees in the designed program. So the problem now is to move the
open cv videos to Unity webcam or video player.

I have little experitise in UDP transmission. Could you give me some
insights on how to achieve this feature( link to other github files is fine
as well).

Thanks,
Francis