hanyazou/TelloPy #14: Can't call additionally created attributes in Tello class in tello.py file

MadelineZhang, 2018-08-01:
When I added an additional attribute to the Tello class and call it in simple_takeoff.py, I got the response that 'Tello' object has no such attribute. 

After trying to print out things in the existing attributes and not getting the desired response, I'm starting to think that there is another Tello class created somewhere else that the simple_takeoff.py function is creating an instance of.

So I guess my questions is: in simple_takeoff.py, is tello.Tello() calling the Tello class in tello.py? If so, why can't I call the additional attributes?

hanyazou, 2018-08-01:
It depends how you install, edit and execute the package to add the attribute. Please show your actual command lines to install, edit and execute the package here.

MadelineZhang, 2018-08-01:
I installed from the source code:
**$ git clone https://github.com/hanyazou/TelloPy 
$ cd TelloPy
$ python3 setup.py bdist_wheel**
although when I try the following line
**$ pip3 install dist/tellopy-0.2.0.dev*.whl --upgrade**
it says it is not a valid wheel filename

I realized that I have also performed **_pip3 install tellopy_** before so I uninstalled it using **_pip3 uninstall tellopy_** so that we only have one version of tellopy - the one that I modified

I edited the code in PyCharm. I added the following attribute under the Tello class
    ****def combined_motion(self, up, down, left, right, forward, back, cw, ccw):
        print('called')
        self.up(up)
        self.down(down)
        self.left(left)
        self.right(right)
        self.forward(forward)
        self.backward(back)
        self.clockwise(cw)
        self.counter_clockwise(ccw)****

Now when I run simpe_takeoff.py using **python3 simple_takeoff.py**, it can no longer import tellopy, so I changed import tellopy to the following:
 **from .. import Tello**
and changed drone = tellopy.Tello() to
**drone = Tello()**
Now when I tried to run it it says:
 _attempted relative import beyond top-level package_

hanyazou, 2018-08-03:
You were running unmodified version which installed with 'pip3 install tellopy'.  That's the why you couldn't call additionally created attributes.

> $ pip3 install dist/tellopy-0.2.0.dev.whl --upgrade*
it says it is not a valid wheel filename

You can specify 'dist/tellopy-\*.dev*.whl ' for the file name instead. (I've fixed the README.md in 770a1c1519b1d22e1e38711ddb2b44c423d120c1)

$ git clone https://github.com/hanyazou/TelloPy 
$ cd TelloPy
$ python setup.py bdist_wheel
$ pip install dist/tellopy-\*.dev*.whl --upgrade

Or you can use --editable option of pip install. Please see 'pip help install'.

$ git clone https://github.com/hanyazou/TelloPy 
$ pip install --editable TelloPy

> python3 simple_takeoff.py

You should specify the package name instead of the file name. (Use -m. Don't use *.py)

$ python -m tellopy.examples.simple_takeoff

closed 2018-08-09 by hanyazou