hanyazou/TelloPy #15: Accessing camera in picture mode

marcollirite, 2018-08-06:
Hi, firstly thanks for this great python-tello interface!

Is there a way to access the Tello's camera in picture mode? I'm able to save frames from the video feed. However, these are lower quality than what is possible in camera mode (at least through the app), and sometimes they capture pixelation from the video stream.

Any help would be greatly appreciated!

hanyazou, 2018-09-01:
@ToxicFrog's fork
https://github.com/ToxicFrog/TelloPy

hanyazou, 2018-09-01:
I found that @ToxicFrog's fork might have the function.

https://github.com/ToxicFrog/TelloPy
https://github.com/ToxicFrog/TelloPy/commit/22f8c95476a92840d08639b752fc86e4fef4d0d6

ToxicFrog, 2018-09-01:
That change (TAKE_PICTURE_COMMAND) is for taking still images and saving them as JPEG. There's other commands for changing camera exposure settings and JPEG compression quality that I haven't added support for yet.

There's a separate commit (https://github.com/ToxicFrog/TelloPy/commit/45b92bceed5c8699aad02f9fa4341cf6b4083181) that adds support for toggling the video stream between "photo mode" (high res 4:3) and "video mode" (low res 16:9); you can capture both photos and videos in either mode.

marcollirite, 2018-09-03:
Thanks for getting back to me. I spoke with ToxicFrog on the Tello forums and have managed to implement my own (worse) version of it in my own environment, but thanks for the solution.

closed 2018-09-04 by hanyazou