hanyazou/TelloPy #25: take_picture() Warn:       file size: payload too small: 00

jaedo5000, 2018-10-28:
Hello,
I would like to know why i can not take a picture. I received the message:

Thank you!

Tello: 13:25:14.920:  Info: unknown packet: 0030 cc 60 00 27 b0 30 00 00 00 00 02 0d
Tello: 13:25:14.921:  Info: recv: file size: cc 90 00 be 88 62 00 2e 00 01 d8 d5 09 00 00 00 2b 5f
Tello: 13:25:14.921:  Info:       file size: num=0 bytes=644568
ALT: 10 | SPD:  0 | BAT: 61 | WIFI: 90 | CAM:  0 | MODE:  6
Tello: 13:25:14.971:  Info: recv: file size: cc 60 00 27 50 62 00 2e 00 00 db ff
Tello: 13:25:14.972:  Warn:       file size: payload too small: 00

This is the code that is almost the same that sample you provide:

from time import sleep
import tellopy

def handler(event, sender, data, **args):
    drone = sender
    if event is drone.EVENT_FLIGHT_DATA:
        print(data)

def test():
    drone = tellopy.Tello()
    try:
        drone.subscribe(drone.EVENT_FLIGHT_DATA, handler)

        drone.connect()
        drone.wait_for_connection(60.0)
        drone.takeoff()
        sleep(5)
        drone.take_picture()
        sleep(5)
        drone.land()
    except Exception as ex:
        print(ex)
    finally:
        drone.quit()

if __name__ == '__main__':
    test()

hanyazou, 2018-11-24:
That message is just a warning and you received some amount of data from your tello.
You should handle the EVENT_FILE_RECEIVED in your program. Please refer the examples/keyboard_and_video.py.

```
def handleFileReceived(event, sender, data):
    global date_fmt
    # Create a file in ~/Pictures/ to receive image data from the drone.
    path = '%s/Pictures/tello-%s.jpeg' % (
        os.getenv('HOME'),
        datetime.datetime.now().strftime('%Y-%m-%d_%H%M%S'))
    with open(path, 'wb') as fd:
        fd.write(data)
    status_print('Saved photo to %s' % path)
```
```
drone.subscribe(drone.EVENT_FILE_RECEIVED,  handleFileReceived)
```

jaedo5000, 2018-11-25:
Hello Hanyazou,
Thank You for your help. Have a good day.

Regards,
Jim.