hanyazou/TelloPy #26: video_effect.py sample code not working

leon6827, 2018-10-31:
Hi,

I noticed that av.open(drone.get_video_stream()) never returns.
Can anyone point out potential area to debug this?

```
        print("Connect Done")
        drone.wait_for_connection(60.0)
        print("Wait for Connect Done")
        container = av.open(drone.get_video_stream())
        print("AV Open Done")
```
Logs attached:

Tello: 12:13:03.980:  Info: start video thread
Tello: 12:13:03.980:  Info: send connection request (cmd="conn_req:9617")
Tello: 12:13:03.981:  Info: video receive buffer size = 524288
Tello: 12:13:03.981:  Info: state transit State::disconnected -> State::connecting
===Connect Done
Tello: 12:13:04.021:  Info: connected. (port=9617)
Tello: 12:13:04.021:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 12:13:04.021:  Info: state transit State::connecting -> State::connected
===Wait for Connect Done
Tello: 12:13:04.055:  Info: get video stream
Tello: 12:13:04.056:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:13:04.072:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 12:13:04.073:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 12:13:04.073:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 12:13:04.073:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 12:13:07.013:  Info: video data 789925 bytes 379.9KB/sec
Tello: 12:13:09.015:  Info: video data 755506 bytes 368.4KB/sec
Tello: 12:13:11.015:  Info: video data 753945 bytes 368.1KB/sec
Tello: 12:13:13.015:  Info: video data 752572 bytes 367.4KB/sec loss=1
Tello: 12:13:14.971:  Info: video recv: 1460 bytes 5200 +784
Tello: 12:13:15.031:  Info: video data 454465 bytes 220.1KB/sec loss=207
Tello: 12:13:16.759:  Info: video recv: 1460 bytes 8300 +228
Tello: 12:13:16.983:  Info: video recv: 1460 bytes 8a01 +105
Tello: 12:13:17.277:  Info: video recv: 1460 bytes 8a06 +290
Tello: 12:13:17.277:  Info: video data 590056 bytes 256.6KB/sec loss=58
Tello: 12:13:17.591:  Info: video recv: 1460 bytes 9a07 +270
Tello: 12:13:17.764:  Info: video recv: 1460 bytes a600 +145
Tello: 12:13:18.083:  Info: video recv: 1460 bytes aa04 +279
Tello: 12:13:18.258:  Info: video recv: 1460 bytes b404 +143
Tello: 12:13:18.563:  Info: video recv: 1460 bytes b904 +274
Tello: 12:13:18.748:  Info: video recv: 1460 bytes be07 +146
Tello: 12:13:18.920:  Info: video recv: 1460 bytes c900 +140
Tello: 12:13:19.067:  Info: video recv: 1460 bytes c901 +147
Tello: 12:13:19.243:  Info: video recv: 1460 bytes d100 +144
Tello: 12:13:19.563:  Info: video recv: 1460 bytes d703 +287
Tello: 12:13:19.563:  Info: video data 188107 bytes  80.3KB/sec loss=538
Tello: 12:13:19.742:  Info: video recv: 1460 bytes e005 +141
Tello: 12:13:19.988:  Info: video recv: 1460 bytes e803 +215
Tello: 12:13:20.163:  Info: video recv: 1460 bytes ef03 +147
Tello: 12:13:21.570:  Info: video data 587037 bytes 285.6KB/sec loss=196
Tello: 12:13:23.151:  Info: video recv: 1460 bytes 4a02 +177
Tello: 12:13:23.337:  Info: video recv: 1460 bytes 4b00 +165
Tello: 12:13:23.668:  Info: video recv: 1460 bytes 5903 +165
Tello: 12:13:23.669:  Info: video data 568066 bytes 264.4KB/sec loss=108
Tello: 12:13:24.019:  Info: video recv: 1460 bytes 6800 +141
Tello: 12:13:24.197:  Info: video recv: 1460 bytes 6d00 +152
Tello: 12:13:24.630:  Info: video recv: 1460 bytes 7600 +271
Tello: 12:13:24.814:  Info: video recv: 1460 bytes 7f04 +140
Tello: 12:13:24.994:  Info: video recv: 1460 bytes 8003 +164
Tello: 12:13:25.160:  Info: video recv: 1460 bytes 8a00 +142
Tello: 12:13:25.332:  Info: video recv: 1460 bytes 8f00 +134
Tello: 12:13:25.651:  Info: video recv: 1460 bytes 9400 +278
Tello: 12:13:25.677:  Info: video data 254564 bytes 123.8KB/sec loss=348
Tello: 12:13:25.828:  Info: video recv: 1460 bytes 9406 +145
Tello: 12:13:26.005:  Info: video recv: 1460 bytes a407 +147
Tello: 12:13:26.185:  Info: video recv: 1460 bytes aa00 +164
Tello: 12:13:26.356:  Info: video recv: 1460 bytes ab03 +145
Tello: 12:13:26.519:  Info: video recv: 1460 bytes b500 +138
Tello: 12:13:26.661:  Info: video recv: 1460 bytes b501 +141
Tello: 12:13:26.843:  Info: video recv: 1460 bytes bd03 +153

clydemcqueen, 2018-10-31:
Try `container = av.container.open(self._drone.get_video_stream())`

leon6827, 2018-10-31:
Thanks for the suggestion, but I saw the same log

leon6827, 2018-10-31:
PyAV v6.0.0
git origin: git@github.com:mikeboers/PyAV
git commit: unknown-commit
library configuration: --prefix=/usr/local/Cellar/ffmpeg/4.0.2_1 --enable-shared --enable-pthreads --enable-version3 --enable-hardcoded-tables --enable-avresample --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gpl --enable-libmp3lame --enable-libopus --enable-libsnappy --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-lzma --enable-opencl --enable-videotoolbox
library license: GPL version 3 or later
libavcodec     58. 18.100
libavdevice    58.  3.100
libavfilter     7. 16.100
libavformat    58. 12.100
libavutil      56. 14.100
libswresample   3.  1.100
libswscale      5.  1.100

leon6827, 2018-11-01:
Seems to be a problem with my busy WiFi environment.
In my case, video_data.gap(self.prev_video_data) always >0, then no data will be appended to the queue. This seems cause av.open() to wait forever.

Now why we have this logic? I should have the option to view my video even though it is choppy right?
 
```
       elif event is self.drone.EVENT_VIDEO_DATA:
            self.log.debug('%s.handle_event(VIDEO_DATA, size=%d)' % (self.name, len(data)))
            video_data = VideoData(data)
            if 0 < video_data.gap(self.prev_video_data):
                self.wait_first_packet_in_frame = True
                
            self.prev_video_data = video_data
            if self.wait_first_packet_in_frame and data[1] != 0:
                self.ignore_packets += 1
                return
            if self.wait_first_packet_in_frame:
                self.log.debug('%s.handle_event(VIDEO_DATA): ignore %d packets' %
                               (self.name, self.ignore_packets))
            self.ignore_packets = 0
            self.wait_first_packet_in_frame = False

            self.cond.acquire()
            self.queue.append(data[2:])
            self.cond.notifyAll()
            self.cond.release()

```

closed 2018-12-08 by hanyazou