hanyazou/TelloPy #29: Tello: 13:10:59.961: Error: video recv: timeout

mariofraga, 2018-11-15:
Hello, congratulations for the Tello initiative.

I'm having trouble trying to use these are some of the errors

Error Running joystick_and_video.py

Tello video WID: 722040
Tello: 13:10:53.959:  Info: start video thread
Tello: 13:10:53.959:  Info: send connection request (cmd="conn_req:9617")
Tello: 13:10:53.959:  Info: video receive buffer size = 524288
Tello: 13:10:53.960:  Info: state transit State::disconnected -> State::connecting
Tello: 13:10:53.960:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 13:10:53.963:  Info: connected. (port=9617)
Tello: 13:10:53.963:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 13:10:53.964:  Info: state transit State::connecting -> State::connected
Tello: 13:10:53.964:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 13:10:53.964:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 13:10:54.017:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 13:10:54.018:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 13:10:59.961: Error: video recv: timeout

Error Running joystick_and_video.py 

Tello: 13:14:19.936:  Info: start video thread
Tello: 13:14:19.937:  Info: send connection request (cmd="conn_req:9617")
Tello: 13:14:19.937:  Info: video receive buffer size = 524288
Tello: 13:14:19.937:  Info: state transit State::disconnected -> State::connecting
Tello: 13:14:19.937:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 13:14:19.977:  Info: connected. (port=9617)
Tello: 13:14:19.977:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 13:14:19.977:  Info: state transit State::connecting -> State::connected
Tello: 13:14:19.978:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 13:14:19.978:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
ALT:  0 | SPD:  0 | BAT:  9 | WIFI:  0 | CAM:  0 | MODE:  1
Tello: 13:14:20.029:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 13:14:20.029:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
ALT:  0 | SPD:  0 | BAT:  9 | WIFI: 90 | CAM:  0 | MODE:  1
Tello: 13:14:25.940: Error: video recv: timeout
Tello: 13:14:30.941: Error: video recv: timeout
Tello: 13:14:35.941: Error: video recv: timeout
Tello: 13:14:40.942: Error: video recv: timeout
Tello: 13:14:45.943: Error: video recv: timeout
ALT:  0 | SPD:  0 | BAT:  8 | WIFI: 90 | CAM:  0 | MODE:  1
Tello: 13:14:50.943: Error: video recv: timeout
Tello: 13:14:55.944: Error: video recv: timeout
Tello: 13:15:00.944: Error: video recv: timeout
Tello: 13:15:05.945: Error: video recv: timeout

AmirLevine02, 2018-11-20:
Hello, 
I am seeing a similar error trying to run the video_effect.py script: 

Amirs-MBP-4:TelloPy-0.5.0 Amir$ python -m tellopy.examples.video_effect
Tello: 21:26:46.484:  Info: start video thread
Tello: 21:26:46.484:  Info: video receive buffer size = 524288
Tello: 21:26:46.484:  Info: send connection request (cmd="conn_req:9617")
Tello: 21:26:46.485:  Info: state transit State::disconnected -> State::connecting
Tello: 21:26:46.492:  Info: connected. (port=9617)
Tello: 21:26:46.492:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 21:26:46.492:  Info: state transit State::connecting -> State::connected
Tello: 21:26:46.493:  Info: get video stream
Tello: 21:26:46.493:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 21:26:46.545:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 21:26:46.546:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 21:26:46.546:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 21:26:46.547:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 21:26:52.491: Error: video recv: timeout
Tello: 21:26:57.496: Error: video recv: timeout
^CTraceback (most recent call last):
  File "/Users/Amir/Documents/MAE148/TelloPy-0.5.0/tellopy/_internal/video_stream.py", line 19, in read
    self.cond.wait(5.0)
  File "/Users/Amir/anaconda3/lib/python3.6/threading.py", line 299, in wait
    gotit = waiter.acquire(True, timeout)
KeyboardInterrupt

For me, the error comes in the following line of code: 

        container = av.open(drone.get_video_stream())

zweberre, 2018-11-30:
I'm receiving the same error in both keyboard_and_video.py and video_effect.py.

However, everything else seems to be working including receiving and saving images.

pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Tello video WID: 6291469
Tello: 07:04:46.235:  Info: start video thread
Tello: 07:04:46.236:  Info: send connection request (cmd="conn_req:9617")
Tello: 07:04:46.236:  Info: video receive buffer size = 425984
Tello: 07:04:46.236:  Info: state transit State::disconnected -> State::connecting
Tello: 07:04:46.236:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 07:04:46.266:  Info: connected. (port=9617)
Tello: 07:04:46.266:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 07:04:46.267:  Info: state transit State::connecting -> State::connected
Tello: 07:04:46.267:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 07:04:46.268:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 07:04:46.318:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 07:04:46.319:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 07:04:52.243: Error: video recv: timeout
Tello: 07:04:57.249: Error: video recv: timeout
Tello: 07:05:02.253: Error: video recv: timeout

PhilsLab, 2018-12-10:
Same for me, no firewall, direct connection -> video recv: timeout
Maybe something has changed in the protocol? The app seems to work fine.

PhilsLab, 2018-12-10:
Seems like "streamon" via the "SDK" always results in an `ERROR` response, is this maybe an issue with a new firmware?

hanyazou, 2018-12-22:
I updated my Tello with newest firmware 01.04.78.1. It can stream video data without any errors.

[image]

CrashxZ, 2020-02-18:
Hi, I am trying to get the video from tello, and I am still getting this error (  Error: video recv: timeout)

mainak@mainak:~$ python -m tellopy.examples.video_effect
Tello: 02:13:07.741:  Info: send connection request (cmd="conn_req:9617")
Tello: 02:13:07.742:  Info: state transit State::disconnected -> State::connecting
Tello: 02:13:07.742:  Info: start video thread
Tello: 02:13:07.742:  Info: video receive buffer size = 425984
Tello: 02:13:07.771:  Info: connected. (port=9617)
Tello: 02:13:07.771:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 02:13:07.772:  Info: state transit State::connecting -> State::connected
Tello: 02:13:07.775:  Info: get video stream
Tello: 02:13:07.775:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 02:13:07.821:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 02:13:07.821:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 02:13:07.822:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 02:13:07.823:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 02:13:13.745: Error: video recv: timeout
Tello: 02:13:18.750: Error: video recv: timeout
Tello: 02:13:23.756: Error: video recv: timeout
Tello: 02:13:28.769: Error: video recv: timeout
Tello: 02:13:33.769: Error: video recv: timeout
Traceback (most recent call last):
  File "/home/mainak/.local/lib/python2.7/site-packages/tellopy/examples/video_effect.py", line 17, in main
    container = av.open(drone.get_video_stream())
  File "av/container/core.pyx", line 274, in av.container.core.open
  File "av/container/core.pyx", line 226, in av.container.core.Container.__cinit__
  File "av/container/core.pyx", line 123, in av.container.core.ContainerProxy.__init__
  File "av/container/core.pyx", line 194, in av.container.core.ContainerProxy.err_check
  File "av/utils.pyx", line 105, in av.utils.err_check
AVError: [Errno 1094995529] Invalid data found when processing input: '<none>'
[Errno 1094995529] Invalid data found when processing input: '<none>'
Tello: 02:13:37.790:  Info: quit
Tello: 02:13:37.790:  Info: state transit State::connected -> State::quit
<class 'tellopy._internal.video_stream.VideoStream'>.handle_event(DISCONNECTED)
Tello: 02:13:37.843:  Info: exit from the recv thread.
Tello: 02:13:38.771: Error: video recv: timeout
Tello: 02:13:38.771:  Info: exit from the video thread.