hanyazou/TelloPy #31: No handlers could be found for logger "libav.h264

dsvilarkovic, 2018-11-22:
I seem to get this while running video_effect,"No handlers could be found for logger "libav.h264". It continues to show canny and imshow window, and every log on terminal is regular one, but those just freeze after first picture is shown.

hanyazou, 2018-11-24:
Could you try the latest version with pip install --upgrade tellopy ?

dsvilarkovic, 2018-11-24:
Yeah, I tried, now it works!

hanyazou, 2018-11-24:
fixed in 6522883c50a60fe217793b5caf76ab296dbf0e6c and 7749936bd8d5d873008c34d9eb386c2c8fe2ddf7

closed 2018-11-24 by hanyazou

arthur-yh, 2019-02-25:
I have the same problem where it shows ""No handlers could be found for logger "libav.h264" when I run `python -m tellopy.examples.video_effect` . 
And there shows 
** 
    No handlers could be found for logger "libav.h264"
    Tello: 16:49:44.744:  Info: LogData: UNHANDLED LOG DATA: id= 1002, length=  27
    Tello: 16:49:45.747:  Info: video data 998945 bytes 487.0KB/sec loss=111
    Tello: 16:49:47.748:  Info: video data 746009 bytes 364.2KB/sec
    Tello: 16:49:49.750:  Info: video data 756175 bytes 368.7KB/sec
    Tello: 16:49:51.751:  Info: video data 752671 bytes 367.5KB/sec
    Tello: 16:49:53.754:  Info: video data 750915 bytes 366.0KB/sec
    ASSERT: "false" in file qasciikey.cpp, line 501
    Aborted (core dumped)
    yhh@elab204d:~/TelloPy$_
**
I have triede ` pip install --upgrad tellopy` there shows
**
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
The directory '/home/yhh/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/yhh/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: tellopy in /usr/local/lib/python2.7/dist-packages (0.7.0.dev0)
You are using pip version 19.0.2, however version 19.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

**
Does it means  I should use the develop_0.6.0 release not the 0.7.0 release ? 
could you help me ? Thank you.

hanyazou, 2019-02-28:
Output message No handlers could be found for logger "libav.h264" is not an error.
In this case, I think ASSERT: "false" in file qasciikey.cpp, line 501 is the error message.
That is not the same error.  You don't have the same problem.