hanyazou/TelloPy #32: Error on install OSX 10.13

ifuchs, 2018-11-23:
I am trying to follow the examples in the readme and I get the following errors

Computer-2:~ ihf$ pip install av
Collecting av
  Downloading https://files.pythonhosted.org/packages/15/80/edc9e110b2896ebe16863051e68bd4786efeda71ce94b81a048d146062cc/av-6.1.0.tar.gz (2.0MB)
    100% |################################| 2.0MB 6.9MB/s
Building wheels for collected packages: av
  Running setup.py bdist_wheel for av ... error
  Complete output from command /Users/ihf/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-install-YLV_Ug/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-wheel-jZqY1E --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.6-x86_64-2.7
  creating build/lib.macosx-10.6-x86_64-2.7/av
  copying av/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av
  copying av/__main__.py -> build/lib.macosx-10.6-x86_64-2.7/av
  copying av/datasets.py -> build/lib.macosx-10.6-x86_64-2.7/av
  copying av/deprecation.py -> build/lib.macosx-10.6-x86_64-2.7/av
  creating build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/audio.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/audio_player.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/average.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/cctx_decode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/cctx_encode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/decode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/decode_threads.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/dump_format.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/encode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/encode_frames.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/experimental.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/filmstrip.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/frame_seek_example.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/glproxy.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/graph.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/merge-filmstrip.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/player.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/qtproxy.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/remux.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/resource_use.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/save_subtitles.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/second_seek_example.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/seekmany.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/show_frames_opencv.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  creating build/lib.macosx-10.6-x86_64-2.7/av/audio
  copying av/audio/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/audio
  creating build/lib.macosx-10.6-x86_64-2.7/av/codec
  copying av/codec/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/codec
  creating build/lib.macosx-10.6-x86_64-2.7/av/container
  copying av/container/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/container
  creating build/lib.macosx-10.6-x86_64-2.7/av/data
  copying av/data/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/data
  creating build/lib.macosx-10.6-x86_64-2.7/av/filter
  copying av/filter/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/filter
  creating build/lib.macosx-10.6-x86_64-2.7/av/subtitles
  copying av/subtitles/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/subtitles
  creating build/lib.macosx-10.6-x86_64-2.7/av/video
  copying av/video/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/video
  running build_ext
  running config
  Could not find libavformat with pkg-config.
  Could not find libavcodec with pkg-config.
  Could not find libavdevice with pkg-config.
  Could not find libavutil with pkg-config.
  Could not find libavfilter with pkg-config.
  Could not find libswscale with pkg-config.
  Could not find libswresample with pkg-config.

  ----------------------------------------
[... 43 lines cut ...]
    copying av/audio/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/audio
    creating build/lib.macosx-10.6-x86_64-2.7/av/codec
    copying av/codec/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/codec
    creating build/lib.macosx-10.6-x86_64-2.7/av/container
    copying av/container/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/container
    creating build/lib.macosx-10.6-x86_64-2.7/av/data
    copying av/data/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/data
    creating build/lib.macosx-10.6-x86_64-2.7/av/filter
    copying av/filter/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/filter
    creating build/lib.macosx-10.6-x86_64-2.7/av/subtitles
    copying av/subtitles/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/subtitles
    creating build/lib.macosx-10.6-x86_64-2.7/av/video
    copying av/video/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/video
    running build_ext
    running config
    Could not find libavformat with pkg-config.
    Could not find libavcodec with pkg-config.
    Could not find libavdevice with pkg-config.
    Could not find libavutil with pkg-config.
    Could not find libavfilter with pkg-config.
    Could not find libswscale with pkg-config.
    Could not find libswresample with pkg-config.

    ----------------------------------------
Command "/Users/ihf/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-install-YLV_Ug/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-record-bdS81Q/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-install-YLV_Ug/av/:

ifuchs, 2018-11-24:
I have never had any problems using pip with anaconda python. Of course, that does not mean the problem is not related to my setup, I just don;t know what needs to change.

hanyazou, 2018-11-25:
Okay, I've newly installed macOS 10.13 onto virtual machine with VMware fusion and installed tools and libraries step by step. Here are all steps that were needed to run the example.
I guess you can do 'brew install ffmpeg' to solve your problem.

```
$ xcode-select --install
$ ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
$ brew install python2
🍺  /usr/local/Cellar/python@2/2.7.15_1: 4,665 files, 82.6MB
$ brew install pkg-config
🍺  /usr/local/Cellar/pkg-config/0.29.2: 11 files, 627.2KB
$ brew install ffmpeg
🍺  /usr/local/Cellar/ffmpeg/4.1: 282 files, 55.4MB
$ pip install av
Successfully installed av-6.1.0
$ pip install tellopy
Successfully installed tellopy-0.6.0
$ python -m tellopy.examples.video_effect
```

ifuchs, 2018-11-25:
The 'brew install ffmpeg' worked but 'pip install av' still fails:
$ pip install av
Collecting av
  Using cached https://files.pythonhosted.org/packages/15/80/edc9e110b2896ebe16863051e68bd4786efeda71ce94b81a048d146062cc/av-6.1.0.tar.gz
Building wheels for collected packages: av
  Running setup.py bdist_wheel for av ... error
  Complete output from command /Users/ihf/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-install-JQt12k/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-wheel-QQGufs --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.6-x86_64-2.7
  creating build/lib.macosx-10.6-x86_64-2.7/av
  copying av/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av
  copying av/__main__.py -> build/lib.macosx-10.6-x86_64-2.7/av
  copying av/datasets.py -> build/lib.macosx-10.6-x86_64-2.7/av
  copying av/deprecation.py -> build/lib.macosx-10.6-x86_64-2.7/av
  creating build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/audio.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/audio_player.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/average.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/cctx_decode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/cctx_encode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/decode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/decode_threads.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/dump_format.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/encode.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/encode_frames.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/experimental.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/filmstrip.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/frame_seek_example.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/glproxy.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/graph.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/merge-filmstrip.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/player.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/qtproxy.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/remux.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/resource_use.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/save_subtitles.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/second_seek_example.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/seekmany.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  copying scratchpad/show_frames_opencv.py -> build/lib.macosx-10.6-x86_64-2.7/scratchpad
  creating build/lib.macosx-10.6-x86_64-2.7/av/audio
  copying av/audio/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/audio
  creating build/lib.macosx-10.6-x86_64-2.7/av/codec
  copying av/codec/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/codec
  creating build/lib.macosx-10.6-x86_64-2.7/av/container
  copying av/container/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/container
  creating build/lib.macosx-10.6-x86_64-2.7/av/data
  copying av/data/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/data
  creating build/lib.macosx-10.6-x86_64-2.7/av/filter
  copying av/filter/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/filter
  creating build/lib.macosx-10.6-x86_64-2.7/av/subtitles
  copying av/subtitles/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/subtitles
  creating build/lib.macosx-10.6-x86_64-2.7/av/video
  copying av/video/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/video
  running build_ext
  running config
  Could not find libavformat with pkg-config.
  Could not find libavcodec with pkg-config.
  Could not find libavdevice with pkg-config.
  Could not find libavutil with pkg-config.
  Could not find libavfilter with pkg-config.
  Could not find libswscale with pkg-config.
  Could not find libswresample with pkg-config.

  ----------------------------------------
  Failed building wheel for av
  Running setup.py clean for av
[... 79 lines cut ...]
    creating build/lib.macosx-10.6-x86_64-2.7/av/codec
    copying av/codec/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/codec
    creating build/lib.macosx-10.6-x86_64-2.7/av/container
    copying av/container/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/container
    creating build/lib.macosx-10.6-x86_64-2.7/av/data
    copying av/data/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/data
    creating build/lib.macosx-10.6-x86_64-2.7/av/filter
    copying av/filter/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/filter
    creating build/lib.macosx-10.6-x86_64-2.7/av/subtitles
    copying av/subtitles/__init__.py -> build/lib.macosx-10.6-x86_64-2.7/av/subtitles
    creating build/lib.macosx-10.6-x86_64-2.7/av/video
[... cut ...]
Command "/Users/ihf/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-install-JQt12k/av/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-record-KbqTxD/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/y2/38gkkc0s7d52_mr_s6zfj7m80000gq/T/pip-install-JQt12k/av/
$

hanyazou, 2018-11-25:
Post output of following commands here.

$ brew install python2
$ brew install pkg-config
$ brew install ffmpeg
$ which python
$ which pip
$ which pkg-config
$ which ffmpeg

hanyazou, 2018-12-08:
Please reopen if you still have the problem.

closed 2018-12-08 by hanyazou