hanyazou/TelloPy #33: Error with v0.6.0 version of keyboard_and_video.py

dsvilarkovic, 2018-11-25:
Last error I got when starting tellopy 0.6.0 is, with usage of:    

> Tello: 01:01:41.501: Error: video recv: [Errno 2] No such file or directory
> Traceback (most recent call last):
>   File "/home/dusan/miniconda2/lib/python2.7/site-packages/tellopy/_internal/tello.py", line 694, in __video_thread
>     self.__publish(event=self.EVENT_VIDEO_FRAME, data=data[2:])
>   File "/home/dusan/miniconda2/lib/python2.7/site-packages/tellopy/_internal/tello.py", line 154, in __publish
>     dispatcher.send(event, sender=self, **args)
>   File "/home/dusan/miniconda2/lib/python2.7/site-packages/tellopy/_internal/dispatcher.py", line 35, in send
>     receiver(event=sig, **named)
>   File "keyboard_and_video.py", line 191, in videoFrameHandler
>     video_player = Popen(cmd + ['-'], stdin=PIPE)
>   File "/home/dusan/miniconda2/lib/python2.7/subprocess.py", line 394, in __init__
>     errread, errwrite)
>   File "/home/dusan/miniconda2/lib/python2.7/subprocess.py", line 1047, in _execute_child
>     raise child_exception
> OSError: [Errno 2] No such file or directory
> 

this happened previously on 0.5.0 , I am not sure where to look at (both on keyboard_and_video.py). Thanks for fast solving previous issues!

hanyazou, 2018-11-25:
In this case, the file which could not find is 'mplayer'  I think. Please just type 'mplayer' on your terminal to check you have mplayer in your system. If it does not exist, you should install mplayer on the execution path.

```
def videoFrameHandler(event, sender, data):
    global video_player
    global video_recorder
    if video_player is None:
        cmd = [ 'mplayer', '-fps', '35', '-really-quiet' ]
        if wid is not None:
            cmd = cmd + [ '-wid', str(wid) ]
        video_player = Popen(cmd + ['-'], stdin=PIPE)
```

closed 2018-12-08 by hanyazou

boahneedle, 2019-05-10:
I'm having the same issue.  Mplayer is installed on my system, and I can run it from terminal with no errors.  

What do you mean by execution path? should i move the MPlayer binaries to the same path as the .py script?

edmondja, 2019-05-23:
NameError: name 'video_player' is not defined
Tello: 00:40:34.601: Error: video recv: name 'video_player' is not defined
Traceback (most recent call last):
  File "/anaconda3/lib/python3.7/site-packages/tellopy/_internal/tello.py", line 694, in __video_thread
    self.__publish(event=self.EVENT_VIDEO_FRAME, data=data[2:])
  File "/anaconda3/lib/python3.7/site-packages/tellopy/_internal/tello.py", line 154, in __publish
    dispatcher.send(event, sender=self, **args)
  File "/anaconda3/lib/python3.7/site-packages/tellopy/_internal/dispatcher.py", line 35, in send
    receiver(event=sig, **named)
  File "crashtest.py", line 25, in handler
    if video_player is None:
NameError: name 'video_player' is not defined

I typed mplayer in my terminal and its present.