hanyazou/TelloPy #36: Drone Swarming Support

ardorem, 2018-12-11:
I guess TelloPy doesn't have support for multiple drones but wanted to confirm with the developers. Is there a way for me to use TelloPy for multiple drones (drone swarming)?

GalBrandwine, 2018-12-11:
Im my opinion it is possible,
But remember, each drone required different wifi network.

So Perhaps multiple wifi cards?

Hope it helps.
Gal

ardorem, 2018-12-11:
Thanks for very fast response! 

So would it work if I just purchase one more tello and wifi adapter and connect them through different network? I worried about the fact that they are using the same ip address (192.168.10.1) and same port number (8889) for connecting to the drone. Wouldn't it be an issue?

GalBrandwine, 2018-12-11:
Hmmm...
A good question.
are you sure it is always the same ip, and port?

Ive found this on the web:

https://tellopilots.com/threads/tello-drone-swarm.288/

Hope it helps.
Gal

GalBrandwine, 2018-12-11:
Im not sure,
Ive been reading carefully the thread from the link above, And it seems to work for them.

It seems weird though, because they have connected same ip and port to different networks. 

Ill ask my professor, and will get back to you..

Good luck!
Gal

ardorem, 2018-12-11:
I think what they do is to put the connection to the completely different networks by manually setting SO_BINDTODEVICE to different values. Since they are on the different network, the same IP/port seem to work. Your professor should have a better understanding, please get back to me! Thanks Gal!

GalBrandwine, 2018-12-11:
I thought the exact same thing (but 100% sure about that).

I do have an idea for testing:
Create a mock that listen to tello's ip and port.

And bound mock to network A,
And Tello to network B, and start sending commands to both networks. if the Tello is reacting only to its conection, than you know fpr sure that the networks are separated.

Hope it helps, and ill get back to you with a proper answer.
Gal

GalBrandwine, 2018-12-11:
Fix:
Not 100% sure about that....

GalBrandwine, 2018-12-11:
Hey there!
Iv'e asked my professor, and it is possible to connect one pc to 2 or more network using a network-card for each network.

So, each Tello is a Hotspot, meaning for each tello you need to establish a different network (wifi dongle), after that you can easily create in one code, a socket per network. 

And it is possible that an ip (and port) will appear on different networks, because ip's are per network.
Hope it helps,
Gal

satnavboy100, 2018-12-22:
I know this is probably a bit late but I have seen a video of some one making a drone swarm with 12 AR drones, to control all of those drones he uses a spare router to give all the drones an IP, I thought I'd mention it because it may be useful.

ardorem, 2018-12-27:
@satnavboy100 Thanks for the pointer! Could you please share the video's link with us if you have?

satnavboy100, 2018-12-27:
@ardorem here https://youtu.be/Hu3p5ZR_i5s I hope it helps give you an idea of how the basis works

thebear132, 2022-10-21:
> Hey there! Iv'e asked my professor, and it is possible to connect one pc to 2 or more network using a network-card for each network.
>
> So, each Tello is a Hotspot, meaning for each tello you need to [...]

For anyone reading this in the future, the Tello drones can be set into _station mode_. This will make them connect to a wifi network, where e.g. all of them then can be accessed using their ip's.

Its the _ap ssid pass_ command on page 5 of the [Tello 2.0 SDK](https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf)