hanyazou/TelloPy #39: av.AVError: [Errno 1094995529] Invalid data found when processing input: 'VideoStream'

JavierYepez, 2019-01-01:
Hello, I'm trying to get the video from another thread. I took the code from your joystick_and_video.py example but I'm getting this:

Tello: 20:27:16.964:  Info: video data 216980 bytes 105.4KB/sec loss=89
Traceback (most recent call last):
  File "D:/MyProjects/Tello/test.py", line 25, in recv_thread
    container = av.open(drone.get_video_stream())
  File "av\container\core.pyx", line 275, in av.container.core.open
  File "av\container\core.pyx", line 227, in av.container.core.Container.__cinit__
  File "av\container\core.pyx", line 129, in av.container.core.ContainerProxy.__init__
  File "av\container\core.pyx", line 195, in av.container.core.ContainerProxy.err_check
  File "av\utils.pyx", line 105, in av.utils.err_check
av.AVError: [Errno 1094995529] Invalid data found when processing input: 'VideoStream'

JavierYepez, 2019-01-18:
I found in this thread from another issue: [https://github.com/hanyazou/TelloPy/issues/18#issuecomment-417926346](https://github.com/hanyazou/TelloPy/issues/18#issuecomment-417926346)

So I tried to modify the code from examples/video_effect.py to avoid the error because in my case the error happens always.
For me in order to avoid the error I had to move "av.open()" to execute it in the same thread as the connection. Someone know why?

Some thing like this:
`drone.connect()`
`drone.wait_for_connection(60.0)`
`container = av.open(drone.get_video_stream())`
`threading.Thread(target=video, args=[container]).start()`

JavierYepez, 2019-01-18:
Yes, here is a gist with the code: [https://gist.github.com/JavierYepez/0fb576360062d0898b968bc5a8794ae2](https://gist.github.com/JavierYepez/0fb576360062d0898b968bc5a8794ae2)

hanyazou, 2019-01-19:
That looks like modified version. Please show the source code which makes the error.

hanyazou, 2019-01-19:
I could reproduce the error by executing the video_effect.py example while it occurs rarely. And I could resolve my problem by retrying to open av stream.
I fixed the video_effect.py in https://github.com/hanyazou/TelloPy/commit/a8897d67988690af5c9ef14f5aac17c044fb62a8

Please try this modification and let me know whether this fix solves your problem on Windows.

JavierYepez, 2019-01-19:
It worked for me! Thanks.

I also want to know why you skip 300 frames at the beginning? And why 300 exactly?

closed 2019-01-19 by JavierYepez