hanyazou/TelloPy #40: Problem with pip install av

XernPLL, 2019-01-10:
I cant install av module. First I installed visual studio and then I got this in the cmd:

`C:\Users\walcz>pip install av
Collecting av
  Using cached https://files.pythonhosted.org/packages/15/80/edc9e110b2896ebe16863051e68bd4786efeda71ce94b81a048d146062cc/av-6.1.0.tar.gz
Building wheels for collected packages: av
  Running setup.py bdist_wheel for av ... error
  Complete output from command c:\python37\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\walcz\\AppData\\Local\\Temp\\pip-install-cwy5vyjv\\av\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d C:\Users\walcz\AppData\Local\Temp\pip-wheel-vd3eg038 --python-tag cp37:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\av
  copying av\datasets.py -> build\lib.win-amd64-3.7\av
  copying av\deprecation.py -> build\lib.win-amd64-3.7\av
  copying av\__init__.py -> build\lib.win-amd64-3.7\av
  copying av\__main__.py -> build\lib.win-amd64-3.7\av
  creating build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\audio.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\audio_player.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\average.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\cctx_decode.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\cctx_encode.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\decode.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\decode_threads.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\dump_format.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\encode.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\encode_frames.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\experimental.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\filmstrip.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\frame_seek_example.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\glproxy.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\graph.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\merge-filmstrip.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\player.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\qtproxy.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\remux.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\resource_use.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\save_subtitles.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\second_seek_example.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\seekmany.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\show_frames_opencv.py -> build\lib.win-amd64-3.7\scratchpad
  copying scratchpad\__init__.py -> build\lib.win-amd64-3.7\scratchpad
  creating build\lib.win-amd64-3.7\av\audio
  copying av\audio\__init__.py -> build\lib.win-amd64-3.7\av\audio
  creating build\lib.win-amd64-3.7\av\codec
  copying av\codec\__init__.py -> build\lib.win-amd64-3.7\av\codec
  creating build\lib.win-amd64-3.7\av\container
  copying av\container\__init__.py -> build\lib.win-amd64-3.7\av\container
  creating build\lib.win-amd64-3.7\av\data
  copying av\data\__init__.py -> build\lib.win-amd64-3.7\av\data
  creating build\lib.win-amd64-3.7\av\filter
  copying av\filter\__init__.py -> build\lib.win-amd64-3.7\av\filter
  creating build\lib.win-amd64-3.7\av\subtitles
  copying av\subtitles\__init__.py -> build\lib.win-amd64-3.7\av\subtitles
  creating build\lib.win-amd64-3.7\av\video
  copying av\video\__init__.py -> build\lib.win-amd64-3.7\av\video
  running build_ext
  running config
  writing build\temp.win-amd64-3.7\Release\include\pyav\config.h
  running cythonize
  Compiling av\buffer.pyx because it depends on c:\python37\lib\site-packages\Cython\Includes\libc\string.pxd.
  [1/1] Cythonizing av\buffer.pyx
  Compiling av\bytesource.pyx because it depends on c:\python37\lib\site-packages\Cython\Includes\cpython\buffer.pxd.
  [1/1] Cythonizing av\bytesource.pyx
  Compiling av\descriptor.pyx because it depends on c:\python37\lib\site-packages\Cython\Includes\libc\stdint.pxd.
  [1/1] Cythonizing av\descriptor.pyx
  Compiling av\dictionary.pyx because it depends on c:\python37\lib\site-packages\Cython\Includes\libc\stdint.pxd.
  [1/1] Cythonizing av\dictionary.pyx
[... 136 lines cut ...]
    creating build\lib.win-amd64-3.7\av\data
    copying av\data\__init__.py -> build\lib.win-amd64-3.7\av\data
    creating build\lib.win-amd64-3.7\av\filter
    copying av\filter\__init__.py -> build\lib.win-amd64-3.7\av\filter
    creating build\lib.win-amd64-3.7\av\subtitles
    copying av\subtitles\__init__.py -> build\lib.win-amd64-3.7\av\subtitles
    creating build\lib.win-amd64-3.7\av\video
    copying av\video\__init__.py -> build\lib.win-amd64-3.7\av\video
    running build_ext
    running config
    writing build\temp.win-amd64-3.7\Release\include\pyav\config.h
    running cythonize
    building 'av.buffer' extension
    creating build\temp.win-amd64-3.7\Release\src
    creating build\temp.win-amd64-3.7\Release\src\av
    C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Dinline=__inline -Iinclude -Ibuild\temp.win-amd64-3.7\Release\include -Ic:\python37\include -Ic:\python37\include -Ic:\python37\include -Ibuild\temp.win-amd64-3.7\Release\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /Tcsrc\av\buffer.c /Fobuild\temp.win-amd64-3.7\Release\src\av\buffer.obj
    buffer.c
    C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:c:\python37\libs /LIBPATH:c:\python37\PCbuild\amd64 /LIBPATH:c:\python37\libs /LIBPATH:c:\python37\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.17763.0\um\x64" avfilter.lib avcodec.lib swscale.lib avdevice.lib swresample.lib avutil.lib avformat.lib /EXPORT:PyInit_buffer build\temp.win-amd64-3.7\Release\src\av\buffer.obj /OUT:build\lib.win-amd64-3.7\av\buffer.cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\src\av\buffer.cp37-win_amd64.lib /OPT:NOREF
    LINK : fatal error LNK1181: nie można otworzyć pliku wejściowego „avfilter.lib”
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1181

    ----------------------------------------
Command "c:\python37\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\walcz\\AppData\\Local\\Temp\\pip-install-cwy5vyjv\\av\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\walcz\AppData\Local\Temp\pip-record-wr1e1tox\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\walcz\AppData\Local\Temp\pip-install-cwy5vyjv\av\`

I have tried with conda, but this made another problem with not found module. I have windows 10

charesredhat, 2019-01-12:
I am facing similar problems from possibly coping  *.lib file into 

Windows Kits installation (C:\Program Files (x86)\Windows Kits\8.1\Lib\winv6.3\um\x64

might work I am following 

https://github.com/joeferner/node-java/issues/424

I did a couple of things to fix it, probably ugly but it's working now :

Installed Node v8.9.1
I copied the file needed from my VS2017 installation (C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.13.26128\lib\x64) to Windows Kits installation (C:\Program Files (x86)\Windows Kits\8.1\Lib\winv6.3\um\x64)
Updating VS2017/2015 C++ Build tools from VS installer
Added VCTargetsPath=C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\v140
Added GYP_MSVS_VERSION=2015
Hope this help 👍

I will update you if it works

charesredhat, 2019-01-12:
ugh I got past it to a new error 

 fatal error C1083: Cannot open include file: 'libavutil/mathematics.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Professional\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX64\\x64\\cl.exe' failed with exit status 2

rsewell97, 2019-01-21:
I lost the will to live after spending 2 days with a mixture of these messages. A friend advised me to try Miniconda and it was working in minutes with conda install av. Hope this helps for others!

kesaroid, 2019-02-05:
I am facing the exact same problem and I have tried everything. Conda install works fine, but installing PyAV through pip and source. I have zero luck with the same exit status 1181. [follow](https://stackoverflow.com/questions/54513325/pip-installation-failure-of-pyav-with-exit-status-1181?noredirect=1#comment95864485_54513325)

kesaroid, 2019-02-05:
Hey @charesredhat, Where did you place the DJi Windows SDK files?

malfonsoNeoris, 2019-12-30:
hi... i have win 10, same problem.
Solved with:

- Download av source. ( git clone)
- Download ffmpeg dev + shared binaries, and unzipped to c:\ffmpeg
- Added ffmpeg bin/lib/include si enviroment path
- python setup.py build and then python setup.py install
- ready to go.

sushansapaliga, 2020-04-04:
@malfonsoNeoris was this on windows ?

malfonso0, 2020-04-05:
yes W10 64 bits
did you manage to solve it?

gemSquared, 2020-04-18:
> hi... i have win 10, same problem.
> Solved with:
> 
> * Download av source. ( git clone)
> * Download ffmpeg dev + shared binaries, and unzipped to c:\ffmpeg
> * Added ffmpeg bin/lib/include si enviroment path
> * python setup.py build and then python setup.py install
> * ready to go.

Hello, I've tried this method but it failed on me on the link.exe part. Errored out 1120.

The thing is, could there be another simpler way of directly feeding video into openCV? Using the SDK mode (sending "command"), I just had to enter the IP and port of the Tello into the VideoCapture. I'm not sure how they work, but the data that comes out of the event subscription looks the same as the data I got from dumping the UDP packets sent in the SDK mode.

malfonso0, 2020-04-18:
hi.. in fact yes.. you can grab tello stream directly with opencv (and adding some option to an enviroment folder).
i have tryed and done it, as you said.. sending commands.
you can send the connect command code, then the start stream
and then create a cv2.capture on a specific ip:port. i beliweve the default is 0.0.0.0:(dont remember)
but.. when starting the tellopy , the default port is changed (or something like that) and a listener is added. some time ago i was trying to "change this" but never did.

let my find the scripts for the commands.. due to covid im working in my personal PC.. but have the tello in home, so i cant try to reproduce

malfonso0, 2020-04-20:
hi, can you try to install pyAV one more time. Here the full instructions
Install AV
Download ffmpeg dev + shared binaries, and unzipped to c:\ffmpeg 
Add ffmpeg bin/lib/include to enviroment path 

$ git clone https://github.com/mikeboers/PyAV.git
$ cd PyAv
$ python setup.py bdist_wheel --ffmpeg-dir=c:\ffmpeg
$ pip install dist/PyAv-*.dev*.whl --upgrade

gemSquared, 2020-04-20:
HEY, people! 😃 I solved all my problems with decoding video more directly using OpenCV!

I got the EVENT_VIDEO_FRAME event data and sent it to a loopback UDP port. Then I just made an openCV VideoCapture thread with the URL "udp://@<insert loopback ip and port>". 

OpenCV just takes the frame data and directly decodes it BY ITSELF! It's a little messy, but no more annoying PyAV install isssues and no mplayer (because installing that on Windows 10 is also hard for me, heh).

gemSquared, 2020-04-20:
HEY, people! 😃 I solved all my problems with decoding video more directly using OpenCV!

I got the EVENT_VIDEO_FRAME event data and sent it to a loopback UDP port. Then I just made an openCV VideoCapture thread with the URL "udp://@insert_ip_and_port".

OpenCV just takes the frame data and directly decodes it BY ITSELF! It's a little messy, but no more annoying PyAV install issues!

@malfonso0 thank you so much for trying to help me, but I quit with that mess ☹️

Also I guess this post doesn't belong here anymore because I didn't install PyAV to get what I want. How would I formally suggest adding this method? It makes getting video from the Tello SO much easier...

malfonso0, 2020-04-20:
great! i was also trying to leave pyav, its quite messy to get the frames that way.
could you send the code of what you did? how you send the data to the udp port?

PS: video of the tello should be like 100 ms delay.. if you have more delay.. search for adding the OPENCV_FFMPEG_CAPTURE_OPTIONS with  nobuffer flag

gemSquared, 2020-04-20:
It's really quick and dirty... also took one of the demo programs as a foundation I think the joystick video one.

First, I define the function that will handle a EVENT_VIDEO_FRAME using the drone.subscribe.

In that handler, I used sockets to send the raw bytes data over a random UDP IP and port.

I then made a separate opencv thread with the VideoCapture and I put in the random IP and port (I used 127.0.0.1:5000) after opening the VideoCapture, I made a conditional while loop that does the cap.read() and imshow stuff (this is where you can have a global variable instead and just use that as frame input for other things) . That's it!

When the program closes be sure to stop the loop in the thread and join() it so it stops correctly.

I'll refine and boil down my code then post it here very soon :smiley:

malfonso0, 2020-04-20:
you can check de imutils videocapture... that does exactly what you said... 
open the capture in a thread and is able to open read  and close.. and manage the thread internally

gemSquared, 2020-04-21:
I got it! Even displays video inside the pygame window! All you need is the "Important Bit". And yes it works better the faster the cap.read() thread loops. So use the global cv2 buffer "vidOut". Python 3 btw.

## Controls: 
- SPACE - quit
- RETURN - JPEG picture in the same folder the file is in (hopefully)
- Z - toggle zoom.

```
import pygame
import time

### vvv IMPORTANT BIT vvv ###
# Minimum you need to get video from tello using tellopy

import socket
import threading
import numpy as np
import cv2
import tellopy

vidOut = None    # Frame that can be used externally

stop_cam = False # Stop flag
cam_error = None # Error message can view or raise()
loopback = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
#loopback.bind(('127.0.0.123',1234))

def cam(): # RUN THE WHILE LOOP AS FAST AS POSSIBLE!
##           VIDEO DISTORTION OTHERWISE! Use vidOut buffer :)
    try:
        global vidOut
        global stop_cam
        global cam_error
        cap = cv2.VideoCapture("udp://@127.0.0.1:5000") # Random address
        if not cap.isOpened:
            cap.open()
        while not stop_cam:
            res, vidOut = cap.read()
    except Exception as e:
        cam_error = e
    finally:
        cap.release()
        print("Video Stream stopped.")

camt = threading.Thread(None,cam) # Start thread
camt.start()

def videoFrameHandler(event, sender, data):   # Video Frame loopback
    loopback.sendto(data,('127.0.0.1',5000)) # Random address

drone = tellopy.Tello()
drone.connect()
drone.start_video()
drone.subscribe(drone.EVENT_VIDEO_FRAME, videoFrameHandler)

### ^^^ IMPORTANT BIT ^^^ ###

### vvv Insignificant test code vvv ###
"""
    Simple stuff. For testing. Add your control code here.
    Right now, only these buttons work:
        SPACE  - Quit
        RETURN - Take JPEG and put in the same folder
                 as this file is in (probably)
        z      - Toggle 4:3 960x720 or 16:9 1080:720

    You can also display video to pygame by doing:
        frame = cv2.cvtColor(vidOut, cv2.COLOR_BGR2RGB)
        frame = np.rot90(frame)
        frame = np.flipud(frame)
        frame = pygame.surfarray.make_surface(frame)
        pygameWindow.blit(frame,(0,0))
    Just make sure the pygame window fits the frame
    or rezise to fit.
"""

def flightDataHandler(event, sender, data):
    #print(data)
    pass

def handleFileReceived(event, sender, data):
    global date_fmt
    # Create a file in the same folder as program (hopefully)
    path = 'tello' + str(int(time.time())) + '.jpeg'
    with open(path, 'wb') as fd:
        fd.write(data)
drone.subscribe(drone.EVENT_FILE_RECEIVED, handleFileReceived)
drone.subscribe(drone.EVENT_FLIGHT_DATA, flightDataHandler)

pygame.init()
pygameWindow = pygame.display.set_mode((1080,720))
pygame.display.set_caption("YAY!")
clock = pygame.time.Clock()

try:
    done = False
    while not done:
        for e in pygame.event.get():
            if e.type == pygame.KEYDOWN:
                if e.key == pygame.K_SPACE:
                    done = True
                if e.key == pygame.K_RETURN:
                    drone.take_picture()
                if e.key == pygame.K_z:
                    drone.set_video_mode(not drone.zoom)
                    
        try:
            frame = cv2.cvtColor(vidOut, cv2.COLOR_BGR2RGB)
            frame = np.rot90(frame)
            frame = np.flipud(frame)
            frame = pygame.surfarray.make_surface(frame)
            pygameWindow.fill((0,0,0))
            pygameWindow.blit(frame,(0,0))
        except:
            pygameWindow.fill((0,0,255))
            
        pygame.display.update()
        clock.tick(30)
finally:
    stop_cam = True
    camt.join()
    drone.quit()
    pygame.quit()
    cv2.destroyAllWindows()
    time.sleep(3)
    print("END")
```

Magnarks, 2024-12-16:
> fatal error C1083: Cannot open include file: 'libavutil/mathematics.h': No such file or directory
> error: command 'C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\cl.exe' failed with exit status 2

Same error try install aiortc