hanyazou/TelloPy #44: Error: mplayer file or directory not found

ericwang82, 2019-02-02:
Hi, 

when I ran the file "keyboard_and_video.py" in the examples folder, an error showed "FileNotFoundError: [Errno 2] No such file or directory: 'mplayer': 'mplayer'". 

However, even I installed the mplayer the result was still the same. 
What is more puzzling is that there are so many kinds of mplayer files on the Internet. I don't know if I got the right one 
The location of the mplayer file could be the reason of the error, yet I am just a novice in programming and have no idea which location should I install at properly.

Thanks for any help.

**the full text of error massage is below:**

Tello: 16:47:37.085:  Info: video recv: 1460 bytes 3502 +350
Tello: 16:47:37.141: Error: video recv: [Errno 2] No such file or directory: 'mplayer': 'mplayer'
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tellopy/_internal/tello.py", line 694, in __video_thread
    self.__publish(event=self.EVENT_VIDEO_FRAME, data=data[2:])
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tellopy/_internal/tello.py", line 154, in __publish
    dispatcher.send(event, sender=self, **args)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tellopy/_internal/dispatcher.py", line 35, in send
    receiver(event=sig, **named)
  File "/Users/ericwang/TelloPy/tellopy/examples/keyboard_and_video.py", line 191, in videoFrameHandler
    video_player = Popen(cmd + ['-'], stdin=PIPE)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'mplayer': 'mplayer'

hanyazou, 2019-02-02:
Additional information is required.
What is the OS and OS version which you are using?
Which mplayer package you installed?
Can you execute mplayer from the command line?

ericwang82, 2019-02-02:
> You need the command line version of mplayer. Try installing it with brew
> using brew install mplayer. After installation, check whether it works
> using the terminal with mplayer. If this results in an command not found
> error or the likes, then restart your terminal and try again.

Hi, Kasireddy

I have installed the command line version of mplayer, and it works when I
execute on Terminal. Yet, the the issue is still exist even I reinstall the
mplayer.

hanyazou, 2019-02-02:
My environment is very similar to yours and I don"t have any issues for mplayer.

```
$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.14.2
BuildVersion:	18C54
$ brew install mplayer
==> Downloading https://homebrew.bintray.com/bottles/mplayer-1.3.0_1.mojave.bott
Already downloaded: /Users/hanyazou/Library/Caches/Homebrew/downloads/98173d42737c0f518726fd5aca22b3455f916fe12d82547987a6f6966290b42a--mplayer-1.3.0_1.mojave.bottle.tar.gz
==> Pouring mplayer-1.3.0_1.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/mplayer/1.3.0_1: 11 files, 27.1MB
$ which mplayer
/usr/local/bin/mplayer

```

ericwang82, 2019-02-02:
May I ask where the TelloPy folder is set on your Mac?

hanyazou, 2019-02-02:
The source code folder is placed in ~/github/TelloPy. I think it does not affect this issue.

hanyazou, 2019-02-06:
Can you try "ktrace" and analyze the output?

ericwang82, 2019-02-06:
sorry have no idea about "ktrace"
could you briefly introduce how to use it?

hanyazou, 2019-02-09:
You can trace target process system calls with some trace tools. I tried "dtruss" on my Mac today.

1. Open two terminals.
2. Before running the keyboard_and_video, ensure that you don't run any other python process. If you have any python process in your system. You should stop the python process.
  ```
  $ ps ax | grep python
  ```
3. On the terminal 1, start type script and dtruss command.
  ```
  $ script dtruss.log
  Script started, output file is typescript
  $ sudo dtruss -f -fn python; exit
  Password:
  dtrace: system integrity protection is on, some features will not be available

          PID/THRD  SYSCALL(args)                  = return
  ```
4. On the terminal 2, start keyboard_and_video example.
  ```
  $ python -m tellopy.examples.keyboard_and_video
  ```

With these steps above, I got 30 thousand output lines in the file dtruss.log. The log contains some information about mplayer. In my case, I can see that python execute mplayer in /usr/local/bin.

```
$ grep mplayer dtruss.log
36377/0x445dfb:  execve("/Users/hanyazou/.platformio/penv/bin/mplayer\0", 0x7F87345A9720, 0x7F8734690000)                = -1 Err#2
36377/0x445dfb:  execve("/usr/local/opt/avr-gcc@7/bin/mplayer\0", 0x7F87345A9720, 0x7F8734690000)                = -1 Err#2
36377/0x445dfb:  execve("/usr/local/Cellar/python@2/2.7.14_3/Frameworks/Python.framework/Versions/2.7/bin/mplayer\0", 0x7F87345EAF50, 0x7F8734690000)            = -1 Err#2
36377/0x445dfb:  execve("/Applications/Arduino-1.8.2.app/Contents/Java/hardware/tools/arm/bin/mplayer\0", 0x7F87345EAF50, 0x7F8734690000)                = -1 Err#2
36377/0x445dfb:  execve("/usr/local/esp-open-sdk/xtensa-lx106-elf/bin/mplayer\0", 0x7F87345EAF50, 0x7F8734690000)                = -1 Err#2
36377/0x445dfb:  execve("~/Library/Android/sdk/platform-tools/mplayer\0", 0x7F87345A9720, 0x7F8734690000)                = -1 Err#2
36377/0x445dfb:  execve("~/go/bin/mplayer\0", 0x7F87345EAF50, 0x7F8734690000)            = -1 Err#2
36377/0x445dfb:  execve("~/bin/mplayer\0", 0x7F87345EAF50, 0x7F8734690000)               = -1 Err#2
36377/0x445e04:  stat64("/usr/local/bin/mplayer\0", 0x7FFEEDF3A1F0, 0x0)                 = 0 0
36377/0x445e04:  open_nocancel("/usr/local/Cellar/mplayer/1.3.0/etc/mplayer/mplayer.conf\0", 0x0, 0x1B6)                 = -1 Err#2
36377/0x445e04:  mkdir("/Users/hanyazou/.mplayer/\0", 0x1FF, 0x0)                = -1 Err#17
36377/0x445e04:  open("/Users/hanyazou/.mplayer/config\0", 0xA01, 0x1B6)                 = -1 Err#17
36377/0x445e04:  open_nocancel("/Users/hanyazou/.mplayer/config\0", 0x0, 0x1B6)          = 7 0
36377/0x445e04:  open_nocancel("/Users/hanyazou/.mplayer/codecs.conf\0", 0x0, 0x1B6)             = -1 Err#2
36377/0x445e04:  open_nocancel("/usr/local/Cellar/mplayer/1.3.0/etc/mplayer/codecs.conf\0", 0x0, 0x1B6)          = -1 Err#2
36377/0x445e04:  open_nocancel("/Users/hanyazou/.mplayer/font/font.desc\0", 0x0, 0x1B6)          = -1 Err#2
36377/0x445e04:  open_nocancel("/usr/local/Cellar/mplayer/1.3.0/share/mplayer/font/font.desc\0", 0x0, 0x1B6)             = -1 Err#2
36377/0x445e04:  open("/Users/hanyazou/.mplayer/input.conf\0", 0x0, 0x0)                 = -1 Err#2
36377/0x445e04:  open("/usr/local/Cellar/mplayer/1.3.0/etc/mplayer/input.conf\0", 0x0, 0x3)              = -1 Err#2
36377/0x445e04:  stat64("/Users/hanyazou/.mplayer/-.conf\0", 0x7FFEEDF38D48, 0x0)                = -1 Err#2
36377/0x445e04:  open_nocancel("/Users/hanyazou/.mplayer/sub/-.ifo\0", 0x0, 0x1B6)               = -1 Err#2
36377/0x445e04:  open_nocancel("/Users/hanyazou/.mplayer/sub/-.idx\0", 0x0, 0x1B6)               = -1 Err#2
36377/0x445e04:  open("/usr/local/bin/mplayer\0", 0x8004, 0x6C692DD2)            = 7 0
36377/0x445e04:  stat64("/usr/local/bin/mplayer\0", 0x7FFEEDF387E0, 0x0)                 = 0 0
36377/0x445e04:  open("/usr/local/bin/mplayer\0", 0x0, 0x1FF)            = 7 0
36377/0x445e04:  getattrlist("/usr/local/bin/mplayer\0", 0x7FFEEDF385F8, 0x7FFEEDF38230)                 = 0 0
36377/0x445e04:  getattrlist("/usr/local/bin/mplayer\0", 0x7FFEEDF37338, 0x7FFEEDF36F70)                 = 0 0
36377/0x445e04:  getattrlist("/usr/local/bin/mplayer\0", 0x7FFEEDF35F00, 0x7FFEEDF36720)                 = 0 0
36377/0x445e04:  getattrlist("/usr/local/Cellar/mplayer/1.3.0/bin/mplayer\0", 0x7FFEEDF35B98, 0x7FFEEDF35A10)            = 0 0
36377/0x445e04:  getattrlist("/usr/local/Cellar/mplayer/1.3.0/bin/mplayer\0", 0x7FFEEDF363D8, 0x7FFEEDF36230)            = 0 0
36377/0x445e04:  getattrlist("/usr/local/bin/mplayer\0", 0x7FFEEDF36350, 0x7FFEEDF36B70)                 = 0 0
36377/0x445e04:  getattrlist("/usr/local/Cellar/mplayer/1.3.0/bin/mplayer\0", 0x7FFEEDF35FE8, 0x7FFEEDF35E60)            = 0 0
36377/0x445e04:  getattrlist("/usr/local/Cellar/mplayer/1.3.0/bin/mplayer\0", 0x7FFEEDF36208, 0x7FFEEDF36080)            = 0 0
36377/0x445e04:  open("/usr/local/Cellar/mplayer/1.3.0/bin/mplayer/..namedfork/rsrc\0", 0x0, 0x1B6)              = -1 Err#2
36377/0x445e04:  lstat64("/usr/local/bin/mplayer\0", 0x7FFEEDF37A70, 0x0)                = 0 0
36377/0x445e04:  open_nocancel("/Users/hanyazou/.mplayer/sub/\0", 0x1100004, 0x0)                = -1 Err#2
```

ericwang82, 2019-02-11:
Thank you 
My result was quite surprising me. When I excuted" $ python -m tellopy.examples.keyboard_and_video, 1.", terminal 2 showed "ImportError: No module named pygame". However, I did install the pygame. I checked it with executing "pip install pygame" and the outcome was "Requirement already satisfied".

The executing result of dtruss:

$	PID/THRD  SYSCALL(args) 		 = return
dtrace: error on enabled probe ID 2366 (ID 906: syscall::thread_selfid:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2411 (ID 816: syscall::issetugid:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2255 (ID 1128: syscall::csrctl:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2569 (ID 500: syscall::csops:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2444 (ID 750: syscall::shared_region_check_np:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2255 (ID 1128: syscall::csrctl:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2166 (ID 630: syscall::getxattr:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2255 (ID 1128: syscall::csrctl:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
dtrace: error on enabled probe ID 2151 (ID 838: syscall::stat64:return): invalid user access in action #5 at DIF offset 0
[... cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/TelloPy-develop-0.7.0/tellopy/examples/keyboard_and_video.py", line 24, in <module>
    import pygame
ImportError: No module named pygame
bash-3.2$ pip3 install pygame
Requirement already satisfied: pygame in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (1.9.4)
You are using pip version 18.1, however version 19.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
bash-3.2$

ericwang82, 2019-02-11:
Good news !! 
I find one loophole of my procedure. I was always using "pip3 install" (I use python3). I just try "pip install" and it comes out that I didn't install pygame of the version. Then I try "python -m tellopy.examples.keyboard_and_video" in Terminal again and it works this time.

However, using IDLE to execute the file is still not working.....

hanyazou, 2019-02-11:
Umm... What is the IDLE?

ericwang82, 2019-02-11:
python IDLE
After installing Python3, uses it to compile Python.

hanyazou, 2019-02-15:
> However, using IDLE to execute the file is still not working.....

Do you get any errors? Please describe your steps using the IDLE and outputs if you fix your trouble.

ericwang82, 2019-02-16:
The error message is the same as my first post.
My steps are clicking right button, selecting IDLE and opening it, then click 'run the module' to run it.
The step shows the error message.

My IDLE version is 3.7.2

hanyazou, 2019-02-16:
Is your first step that clicking right button on the the file "keyboard_and_video.py" in the examples folder? If so, that is the problem.

I wrote simple python script that just show PATH environment variable and run the script with IDLE.
```
import os
print(os.environ['PATH'])
```
And open it with IDLE from the finder.
[image]
[image]

Then I run it.
[image]

The script show the environment executable search path.
```
Python 2.7.15 (default, Jun 17 2018, 12:46:58) 
[GCC 4.2.1 Compatible Apple LLVM 9.1.0 (clang-902.0.39.2)] on darwin
Type "copyright", "credits" or "license()" for more information.
>>> WARNING: The version of Tcl/Tk (8.5.9) in use may be unstable.
Visit http://www.python.org/download/mac/tcltk/ for current information.

=========== RESTART: /private/tmp/untitled folder/show-the-PATH.py ===========
/usr/bin:/bin:/usr/sbin:/sbin
>>> 
```

In this case, my executable search path of my system is "/usr/bin:/usr/sbin:/sbin". This search path does not contains the  mplayer.

Please check this in your environment. But I couldn't find how to change this settings with googling. This output seems to be ignoring /etc/paths and /etc.paths.d configurations. I was not noticed this issue because I always use the terminal app instead of the IDLE.

hanyazou, 2019-02-16:
This post below says that you can add to the search path with editing /Applications/Python\ 2.7/IDLE.app/Contents/Resources/idlemain.py

https://stackoverflow.com/questions/3387695/add-to-python-path-mac-os-x

ericwang82, 2019-03-17:
hi
sorry for late reply, I was quite busy recently to deal with the problem.
thank you for the help of the search path advise that I finally solved the problem.

closed 2019-03-17 by hanyazou