hanyazou/TelloPy #45: I don't know how to get a picture I took.

omihub1996, 2019-02-15:
Thank you for your splendid work.

I'm new to Tello and I can't understand how to take a picture and save it as image file by TelloPy.
I'm using take_picture() method as follows but got a bunch of error messages.
Should I make some variable for the return value of take_picuter() method and then convert it to img file somehow?
Can someone clarify this for me?

Thank you in advance!

```python
import tellopy
tl = tellopy.Tello()
tl.connect()
tl.takeoff()
tl.take_picture()
tl.quit()
```

hanyazou, 2019-02-15:
You should handle EVENT_FILE_RECEIVED.
Please look at tellopy/examples/keyboard_and_video.p to know how to handle the event.

pnehareddy1, 2019-09-09:
Can you please help me with this issue - I think you can solve it since you were able to establish connection: (Thanks)

https://github.com/hanyazou/TelloPy/issues/58#issuecomment-529257754

pnehareddy1, 2019-09-09:
@omihub1996