hanyazou/TelloPy #48: video recv: timeout

inspectred, 2019-04-02:
First of all I love your work and most things work perfectly but when I run anything that uses the camera I get presented with:

This shows the feed I get when using video_effect.py but I have also tested on keyboard_and_video.py and got the same output although I could still move the drone around with the keyboard.

Tello: 18:44:15.707:  Info: get video stream
Tello: 18:44:15.707:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:44:15.805:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:15.805:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 18:44:15.805:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:15.805:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 18:44:16.740:  Info: recv: log_header: id=c835, 'b'BUILD Sep 10 2018 15:58:59''
Tello: 18:44:16.787:  Info: LogData: UNHANDLED LOG DATA: id=32768, length=  49
Tello: 18:44:16.787: Error: LogData: corrupted data at pos=1027, data=
Tello: 18:44:16.787: Error: LogData: corrupted data at pos=0, data=d4 9d 91 de d7 d7 c2 d4 c5 8b 91 81 9d 91 c2 d8 cb d4 8b 91 82 90 bc bb b1 d6 75 55 39 00 44 00 80 0b d6 70 00 2b 2b 2b 2b 2b 2b 3b 2b 50 47 26 59 48 56 79 6e 6a 6f 2b 6d 67 6a 78 63 2b 78 7e 68 68 6e 78 78 27 2b 68 7e 79 2b 7f 72 7b 6e 31 3a 0b 84 c9 55 29 00 a8 00 80 2a d6 70 00 0a 0a 0a 0a 0a 0a 1a 0a 71 66 07 78 69 77 14 19 1c 1e 0a 1b 1a 18 1e 0a 1b 1c 12 1e 2a db 8b 55 29 00 a8 00 80 43 d6 70 00 63 63 63 63 63 63 73 63 18 0f 6e 11 00 1e 7d 70 75 77 63 72 73 71 77 63 72 75 7b 77 43 02 a7 55 29 00 a8 00 80 5d d6 70 00 7d 7d 7d 7d 7d 7d 6d 7d 06 11 70 0f 1e 00 63 6e 6b 69 7d 6c 6d 6f 69 7d 6c 6b 65 69 5d 3c af 55 29 00 a8 00 80 77 d6 70 00 57 57 57 57 57 57 47 57 2c 3b 5a 25 34 2a 49 44 41 43 57 46 47 45 43 57 46 41 4f 43 77 fc 1e 55 3a 00 11 00 80 2c bc 76 00 0c 0c 0c 0c 0c 0c 1c 0c 77 60 01 6f 6a 6b 71 7c 61 16 4d 48 48 0c 5e 43 43 58 0c 5b 43 5e 47 16 0c 5c 41 73 43 42 73 58 45 4f 47 21 26 2c 6d de 55 37 00 98 00 80 87 bc 76 00 a7 a7 a7 a7 a7 a7 b7 a7 dc cb aa d4 c2 c9 d4 c8 d5 da dc c6 d4 d5 da e6 f4 f5 a7 ef e8 f4 f3 a7 f4 fe e9 e4 a7 f4 f3 e6 f5 f3 87 2e dd 55 41 00 84 00 80 a2 bc 76 00 82 82 82 82 82 82 92 82 f9 ee 8f f1 e7 ec f1 ed f0 ff f9 e3 f1 f0 ff d1 db cc c1 82 ca cd d1 d6 82 ce c7 c3 c6 82 cd c4 c4 d1 c7 d6 82 92 da 93 93 c3 af a8 a2 e3 91 55 31 00 32 00 80 4f bf 76 00 6f 6f 6f 6f 6f 6f 7f 6f 14 03 62 1c 0a 01 1c 00 1d 12 6f 27 38 10 39 2a 3d 6f 39 20 23 3b 2e 28 2a 75 6f 7f 4f 4c b7 55 38 00 80 00 80 c7 bf 76 00 e7 e7 e7 e7 e7 e7 f7 e7 9c 8b ea 94 8e 8a 9a b5 a2 a6 a3 e7 b7 a6 b5 a6 aa a2 b3 a2 b5 e7 b4 b2 a4 a4 a2 b4 b4 a1 b2 ab ab be e6 c7 a8 b0 55 4d 00 c9 00 80 7b c0 76 00 5b 5b 5b 5b 5b 5b 4b 5b 20 37 56 3d 36 2e 54 36 34 2f 34 29 26 09 1e 1a 1f 5b 16 14 0f 14 09 5b 1d 14 09 18 1e 5b 1f 12 08 1a 19 17 1e 5b 1d 17 1a 1c 5b 08 0e 18 18 1e 08 08 1d 0e 17 17 02 5a 7b ad 5c 55 2c 00 57 00 80 a3 c0 76 00 83 83 83 83 83 83 93 83 f8 ef 8e f0 e6 ed f0 ec f1 fe ca ce d6 83 c4 d1 cc d6 d3 93 83 cc c8 a3 d9 fe 55 3f 00 ee 00 80 c5 c0 76 00 e5 e5 e5 e5 e5 e5 f5 e5 9e 89 e8 82 9c 97 8a 9a 84 86 86 98 ac a8 b0 9a f5 e5 a2 b7 aa b0 b5 e8 fb b6 a0 ab b6 aa b7 9a ac a1 ff f4 f3 f4 fc f5 f1 e5 c5 21 35 55 47 00 2e 00 80 1e c1 76 00 3e 3e 3e 3e 3e 3e 2e 3e 45 52 33 59 47 4c 51 41 5f 5d 5d 43 79 67 6c 41 7f 7d 7d 41 7d 71 70 78 77 79 30 73 6d 7d 41 6c 7b 6f 6b 77 6c 7b 41 6d 77 7a 7b 24 36 28 2d 37 3e 14 1e 05 b8 55 48 00 36 00 80 37 c1 76 00 17 17 17 17 17 17 07 17 6c 7b 1a 73 75 70 06 6a 54 56 5b 5e 68 44 40 5e 43 54 5f 68 54 58 59 51 5e 50 19 5e 5a 42 68 47 45 58 47 52 5b 5b 52 45 68 54 56 5b 5e 0d 1f 05 1e 17 3d 37 c0 20 55 61 00 45 00 80 dd c1 76 00 fd fd fd fd fd fd ed fd 86 91 f0 9a 84 8f 92 82 9c 9e 9e 80 a9 a8 af b3 fd b2 b3 fd a9 b8 b0 ad fd be a9 af b1 f5 ed f4 f1 fd be a8 af 82 a9 b8 b0 ad e7 f0 e8 ed f3 ed f1 fd af b8 bc ae b2 b3 e7 a9 b8 b0 ad fd be a9 af b1 fd be b2 af b8 fd b4 b3 b4 a9 dd c1 d4 55 57 00 c2 00 80 04 c2 76 00 24 24 24 24 24 24 34 24 5f 48 29 43 5d 56 4b 5b 45 47 47 59 6d 69 71 5b 34 24 6b 68 60 24 69 6d 77 5b 67 be a5
Tello: 18:44:16.787: Error: LogData: corrupted data at pos=0, data=65 68 6d 5b 62 68 65 63 3e 62 62 62 62 24 69 6d 77 5b 67 65 68 6d 5b 66 6b 65 76 60 5b 62 68 65 63 3e 62 62 62 62 24 04 61 02 5b 9c
Tello: 18:44:16.834:  Info: LogData: UNHANDLED LOG DATA: id= 2064, length=  64
Tello: 18:44:16.834:  Info: LogData: UNHANDLED LOG DATA: id= 2208, length=  40
Tello: 18:44:16.834:  Info: LogData: UNHANDLED LOG DATA: id=10086, length=   4
Tello: 18:44:16.834:  Info: LogData: UNHANDLED LOG DATA: id=10085, length=  80
Tello: 18:44:16.834:  Info: LogData: UNHANDLED LOG DATA: id= 1710, length=  38
Tello: 18:44:16.834:  Info: LogData: UNHANDLED LOG DATA: id= 1712, length=  10
Tello: 18:44:16.950:  Info: LogData: UNHANDLED LOG DATA: id=   16, length=   4
Tello: 18:44:16.950:  Info: LogData: UNHANDLED LOG DATA: id=   12, length=  55
Tello: 18:44:16.997:  Info: LogData: UNHANDLED LOG DATA: id= 1000, length=  44
Tello: 18:44:16.997:  Info: LogData: UNHANDLED LOG DATA: id= 1001, length=  16
Tello: 18:44:17.711: Error: video recv: timeout
Tello: 18:44:17.711:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:44:17.711:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:17.711:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 18:44:17.758:  Info: LogData: UNHANDLED LOG DATA: id= 1002, length=  27
Tello: 18:44:17.758:  Info: LogData: UNHANDLED LOG DATA: id=  112, length=  35
Tello: 18:44:18.716: Error: video recv: timeout
Tello: 18:44:18.716:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:44:18.732:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:18.732:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 18:44:19.718: Error: video recv: timeout
Tello: 18:44:19.718:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:44:19.765:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:19.765:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 18:44:20.721: Error: video recv: timeout
Tello: 18:44:20.721:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:44:20.737:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:20.737:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
...
...
...
Tello: 18:44:44.849: Error: video recv: timeout
Tello: 18:44:44.849:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:44:44.871:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:44:44.871:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9

I have found an opened issue with similar errors but since its been closed and mine still isn't working, thought I should open it again.

I have narrowed the problem down to this line : 
                container = av.open(drone.get_video_stream())

Hope someone can help out.
Thanks

pnehareddy1, 2019-09-09:
Can you please help me with this issue @inspectred 

https://github.com/hanyazou/TelloPy/issues/58#issuecomment-529257754

inspectred, 2019-09-17:
@pnehareddy1 did you solve this issue? Sorry I'm seeing your message just now but if you still have the issue comment again and I will go through my code to remind myself how I made it work.

pnehareddy1, 2019-09-17:
Thanks for asking me again. I am still having the same issue. I think the problem is with my 'mplayer' on 'Ubuntu 18.04'.  I am able to play the video from the Terminal by using 'mplayer', but unable to play directly by clicking on the video (able to hear the sound, but unable to view the video). Could you help me with this or can you suggest me any other player that works with 'Tellopy'.

Regards,
Neha Puchakayala

ricafort, 2020-07-06:
I am also having the same errors and issue

Walt-H, 2020-07-29:
I had to watch the video via Chrome, but I was able to fix this issue on Windows (see [comment here](https://github.com/hanyazou/TelloPy/issues/78#issuecomment-660214525))

Walt-H, 2020-11-16:
#90

samlaf, 2021-02-23:
I think I found a solution for this.
You just need to add a call to `wait_for_connection`:
```
drone.connect()
drone.wait_for_connection(10.0)
drone.start_video()
```
Alternatively, you can add the call to `wait_for_connection` directly inside `connect`, as I did in this [pull request](https://github.com/hanyazou/TelloPy/pull/96).

SMRazaRizvi96, 2021-09-02:
I am using ROS Noetic on Ubuntu 20.04 and I am also facing the same issue.
I even added
drone.wait_for_connection(10.0)
But no luck.

Please help.