hanyazou/TelloPy #53: question on units

mattganis, 2019-07-16:
Not sure if this is the correct place to ask, but I'm stumped.

Love this library, but I can't figure out the units for various movements.  so when I issue a left(10) for example, is that cm, inches, etc ?   but more puzzling are the rotations, so I can issue a clockwise(90) but it doesn't seem the value (which says from 1-100) is a degree.

what units are used ?   Again, sorry if this is the wrong place to ask - pointers to the correct place would be appreciated !

-Matt

mattganis, 2019-07-16:
so it looks like clockwise(100) spins the drone in a clockwise direction faster than clockwise(25) - so the value passed is not unit, but a spin "rate" - how would I get it to turn just 90 degrees ?

hanyazou, 2019-07-19:
DJI official SDK seems have the commands which you are looking for.
https://dl-cdn.ryzerobotics.com/downloads/tello/20180910/Tello%20SDK%20Documentation%20EN_1.3.pdf

Unfortunately, the tellopy can not send the commands to the drone because the tellopy switched the drone into other command mode and the drone can not accept the commands. I think there could be some commands which have same functionality in the binary command mode which tellopy are using. All commands currently supported by the tellopy were discovered by the community using some reverse engineering. But I don't know binary commands which have the functionality for now.

I hope some one will find related information and comment here.