hanyazou/TelloPy #55: video_effect.py sample code not working

nakataniyota, 2019-08-05:
hi.

I noticed that container = av.open(drone.get_video_stream()) never returns.
Can anyone point out potential area to debug this?

`drone.connect()
        drone.wait_for_connection(60.0)

        retry = 3
        container = None
        while container is None and 0 < retry:
            retry -= 1
            try:
                container = av.open(drone.get_video_stream())
                #av.container.open(drone.get_video_stream())
                print("OPEN AV")`

log attached:

❯ python video_effect.py
Tello: 18:50:21.026:  Info: start video thread
Tello: 18:50:21.027:  Info: send connection request (cmd="conn_req:9617")
Tello: 18:50:21.027:  Info: video receive buffer size = 524288
Tello: 18:50:21.027:  Info: state transit State::disconnected -> State::connecting
Tello: 18:50:21.166:  Info: connected. (port=9617)
Tello: 18:50:21.167:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 18:50:21.167:  Info: state transit State::connecting -> State::connected
Tello: 18:50:21.168:  Info: get video stream
Tello: 18:50:21.168:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 18:50:21.319:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 18:50:21.319:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 18:50:24.050:  Info: video data 978038 bytes 473.3KB/sec
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
・・・
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
non-existing PPS 0 referenced
non-existing PPS 0 referenced
decode_slice_header error
no frame!
Tello: 18:50:26.050:  Info: video data 752425 bytes 367.4KB/sec
OPEN AV
[1]    59818 segmentation fault  python video_effect.py

OSver:Mac OS X 10.14.5
BuildVersion:	18F132
av:6.2.0

pnehareddy1, 2019-09-09:
@nakataniyota 
Can you please help with the connection problem I am facing since you were able to establish connection:
Here is the issue - https://github.com/hanyazou/TelloPy/issues/59#issue-490709711

ricafort, 2020-07-06:
@hanyazou . I am having same issue too. not a connection problem

ricafort, 2020-07-06:
@hanyazou . I am having same issue too. not a connection problem as i can stream using other python code not using this library

xiepip, 2022-09-29:
I am having the same issue now.
Did you fixed it? Thanks @ricafort