hanyazou/TelloPy #59: Not able to connect

pnehareddy1, 2019-09-08:
neha@neha-Inspiron-7573:~/Desktop/Cloud_Computing$ python -m tellopy.examples.video_effect
Tello: 00:24:35.545:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:35.545:  Info: state transit State::disconnected -> State::connecting
Tello: 00:24:35.545:  Info: start video thread
Tello: 00:24:35.545:  Info: video receive buffer size = 425984
Tello: 00:24:37.547:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:39.550:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:41.552:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:43.555:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:45.556:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:47.559:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:49.562:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:51.563:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:53.564:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:55.567:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:57.570:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:24:59.572:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:01.575:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:03.578:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:05.580:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:07.583:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:09.584:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:11.587:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:13.590:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:15.592:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:17.595:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:19.598:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:21.601:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:23.603:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:25.606:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:27.609:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:29.611:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:31.614:  Info: send connection request (cmd="conn_req:9617")
Tello: 00:25:33.617:  Info: send connection request (cmd="conn_req:9617")
Traceback (most recent call last):
  File "/home/neha/.local/lib/python2.7/site-packages/tellopy/examples/video_effect.py", line 15, in main
    drone.wait_for_connection(60.0)
  File "/home/neha/.local/lib/python2.7/site-packages/tellopy/_internal/tello.py", line 133, in wait_for_connection
    raise error.TelloError('timeout')
TelloError: TelloError::timeout
TelloError::timeout
Tello: 00:25:35.546:  Info: quit
Tello: 00:25:35.546:  Info: state transit State::connecting -> State::quit
Tello: 00:25:35.613:  Info: exit from the video thread.
Tello: 00:25:35.620:  Info: exit from the recv thread.

pnehareddy1, 2019-09-08:
I see "timeout" issue. Can you please help with this - How to establish connection. I am using UBUNTU 18.04.3

hanyazou, 2019-09-09:
Your Tello will be an Wi-FI access point after you turn it on.
You should connect your desktop/laptop to the access point, which name is "TELLO-XXXXXX".
('XXXXXX' part will be some numbers and alphabets. it vary.)
Did you connect to the access point and get some IP address for your PC successfully?

pnehareddy1, 2019-09-09:
I am not connecting it to a Tello yet. I am doing preliminary study - trying to get a picture (or) video first on my laptop screen (Ubuntu).

I followed these steps till now:
1. Installed TelloPy with this command- "pip3 install tellopy"
2. Ran your script "simple_takeoff.py" and got this error (I tried to run other scripts too but ended with the same error)

Please let me know if I am missing any steps here? Thanks for the help

hanyazou, 2019-09-09:
This tello.py software runs with the Tello drone. The error means the software can't talk with any drone because your PC does not have any connection with any drone. You must connect your PC to some Tello drone at first.
Picture and video comes from the camera of the Tello drone. You can't get any picture nor video without the Tello drone.

pnehareddy1, 2019-09-09:
Thank you for the quick reply. I will try with the Tello drone and let you know if I have any further questions.
Thanks again for the help.

samlaf, 2021-02-23:
I think I found a solution for this.
You just need to add a call to `wait_for_connection`:
```
drone.connect()
drone.wait_for_connection(10.0)
drone.start_video()
```
Alternatively, you can add the call to `wait_for_connection` directly inside `connect`, as I did in this [pull request](https://github.com/hanyazou/TelloPy/pull/96).