hanyazou/TelloPy #60: Question and parameters passed

mattganis, 2019-09-13:
I don't see it documented anywhere - but when I used the clockwise(x) method (or any other method that has the drone rotate) - what does the 'x' represent ?   

For the forward(x) method, I see the 'x' represents cm/sec speed - but I can't figure out the rotations

thanks.....Matt

RoyalArdor, 2019-10-15:
If you go to _internal folder then tello.py

```
    def clockwise(self, val):
        """
        Clockwise tells the drone to rotate in a clockwise direction.
        Pass in an int from 0-100.
        """
        log.info('clockwise(val=%d)' % val)
        self.left_x = val / 100.0
```

mattganis, 2020-02-29:
So looking at the code above:  self.left_x = val / 100.0

doe the value passed (val) represent a percent around a circle ?   For example if I pass 50, self.left_x is going to be 0.5 (so is that halfway around ?).

I guess I'm trying to figure out to tell it simply:  Rotate 180 degrees

RoyalArdor, 2020-03-02:
No, I believe it is the speed, you can see it being used in the joystick example in the joystick folder.

You can just do this for a 0-360 discrete turn

command = "cw 180"
try:
    self.sock.sendto(command.encode('utf-8'), self.tello_addr)