hanyazou/TelloPy #64: Video Stream does not work

duncanmix, 2019-09-29:
Pygame shows a black screen when the code runs

trooperisme, 2020-03-07:
same here @hanyazou please come and help us

hanyazou, 2020-03-08:
Some people, including me, can run without errors. But others cannot...
At least, I tested on MacOS 10.13 and ubuntu 16.04.
Are you using Windows?

trooperisme, 2020-03-10:
@hanyazou first of all, I am sorry for the late response. Yes, I actually run on Windows 7 and when I try the other sample script such as "python -m tellopy.examples.joystick_and_video". I ended up getting the error 

> Tello: 01:01:41.501: Error: video recv: [Errno 2] No such file or directory
Traceback (most recent call last):
File "/home/dusan/miniconda2/lib/python2.7/site-packages/tellopy/_internal/tello.py", line 694, in __video_thread
self.__publish(event=self.EVENT_VIDEO_FRAME, data=data[2:])
File "/home/dusan/miniconda2/lib/python2.7/site-packages/tellopy/_internal/tello.py", line 154, in __publish
dispatcher.send(event, sender=self, **args)
File "/home/dusan/miniconda2/lib/python2.7/site-packages/tellopy/_internal/dispatcher.py", line 35, in send
receiver(event=sig, **named)
File "keyboard_and_video.py", line 191, in videoFrameHandler
video_player = Popen(cmd + ['-'], stdin=PIPE)
File "/home/dusan/miniconda2/lib/python2.7/subprocess.py", line 394, in init
errread, errwrite)
File "/home/dusan/miniconda2/lib/python2.7/subprocess.py", line 1047, in _execute_child
raise child_exception
OSError: [Errno 2] No such file or directory
And then I try to find the issue on github, I find out that you have already solved the issue in this thread: https://github.com/hanyazou/TelloPy/issues/33. Actually, I try to follow your instruction (install the latest mplayer from pypi). However, I still get the same issue. 
(I am thinking of getting a brand new laptop to give it another try now lol)

trooperisme, 2020-03-10:
And then I try to find the issue on github, I find out that you have already solved the issue in this thread: https://github.com/hanyazou/TelloPy/issues/33. Actually, I try to follow your instruction (install the latest mplayer from pypi). However, I still get the same issue. 
(I am thinking of getting a brand new laptop to give it another try now lol)

hanyazou, 2020-03-15:
I am not familiar with Windows. Did you check where mplayer executable file and dynamic link libraries are? Did you try to add those path to your environment variables?

trooperisme, 2020-03-19:
hi @hanyazou , could you review https://github.com/hanyazou/TelloPy/issues/78 (its very easy, im just not familiar with '%s')

ricafort, 2020-07-06:
@hanyazou . I am having same issue too

Helge543, 2021-08-19:
> Some people, including me, can run without errors. But others cannot...
> At least, I tested on MacOS 10.13 and ubuntu 16.04.
> Are you using Windows?

what did you do that the PyGame screen shows correct images on Ubuntu ?