hanyazou/TelloPy #72: 'Tello' object has no attribute 'EVENT_LOG_DATA'

GalBrandwine, 2019-12-26:
Hey hanyazou!

for some reason I get an error when trying to subscribe to:
```python
 drone.subscribe(drone.EVENT_LOG_DATA, handler)
```

in the handler itself:
```python
def handler(event, sender, data, **args):
    drone = sender    
    if event is drone.EVENT_FLIGHT_DATA:
        flight_data = data
        print(flight_data)
    elif event is drone.EVENT_LOG_DATA:
        log_data = data
        print(log_data.imu)
```

I get the error:
'Tello' object has no attribute 'EVENT_LOG_DATA'

Any ideas why?
Thanks, Gal.

P.S
it's nice to go back and work with the tello again :)
it reminded me how awesome the TelloPy repo *is*!

benjaminjacobreji, 2020-01-15:
I used EVENT_LOG instead of EVENT_LOG_DATA
it helped me solve issues when doing t[ello-openpose](https://github.com/geaxgx/tello-openpose) project

GalBrandwine, 2020-01-18:
UPDATE:
Tried that:
```python
def log_data_handler(event, sender, data):
        """
            Listener to log data from the drone.
        """  
        pos_x = -data.mvo.pos_x
        pos_y = -data.mvo.pos_y
        pos_z = -data.mvo.pos_z
```

The data received is Bytes - hence cant access to its internals..

Any help there?
How did you accessed to the internals of "data"? - I've seen youre project (quite impressive), what have i missed?

Thanks, Gal

gemSquared, 2020-04-21:
Hello! I'm going to assume you use python 3? I also get a bytes string and I run python 3.6.

gemSquared, 2020-04-21:
Okay! So! Download the latest tellopy (develop-0.7.0 (default)) library from here, not pip.

Then go back and use EVENT_LOG_DATA.

If we print() the data you get a string that looks a little something like:
`MVO: VEL:  0.00  0.00  0.00 POS:  0.02  0.01 -0.00|IMU: ACC: -0.05 -0.02 -1.01 GYRO: -0.01 -0.01  0.00 QUATERNION:  1.00  0.01 -0.02 -0.01 VG: -0.00 -0.00  0.00`

Now we just need to parse this. (and I don't know how lol, pls help?) 
Anyway, this should get you closer!

gemSquared, 2020-04-21:
Also seems the height and battery data are in the EVENT_FLIGHT_DATA according to
https://tellopilots.com/threads/imu-state-data.2119/#post-9403

The "POS" data likes to jump to zero when the tello lands and goes back when it takes off. (height is inverted). The "QUATERNION" contains the rotation data (probably want to turn it into Euler first then extract yaw)

lmmedino, 2020-05-01:
Hi, I am still confused on how to get the position information @gemSquared, can you explain further in what part we are getting the x,y,z positions?

gemSquared, 2020-05-02:
The EVENT_LOG_DATA event passes what seems to be a class. Here's what I did to get the container:
```
sensors = None
def logDataHandler(event, sender, data):
    global sensors
    sensors = data

drone.subscribe(drone.EVENT_LOG_DATA, logDataHandler)
```
Here's where I get the data out, for example yaw (you might need this btw)
```
def quaternion2yaw(q0,q1,q2,q3):
    siny = 2.0*(q0*q3+q1*q2)
    cosy = 1.0-2.0*(q2*q2+q3*q3)
    return math.atan2(siny,cosy)

yaw = quaternion2yaw(
                sensors.imu.q0,
                sensors.imu.q1,
                sensors.imu.q2,
                sensors.imu.q3)
```

The format is:

EVENT_LOG_DATA
  - mvo
    - pos_x
    - pos_y
    - pos_z (I used EVENT_FLIGHT_DATA... might try this again)
    - vel_x
    - vel_y
    - vel_z
  - imu
    - acc_x
    - acc_y
    - acc_z
    - gyro_x
    - gyro_y
    - gyro_z
    - q0 (Quaternion used to store orientation. Is weird but it works great)
    - q1
    - q2
    - q3
    - vg_x
    - vg_y
    - vg_z

lmmedino, 2020-05-03:
What do you mean by "container"?

gemSquared, 2020-05-04:
I don't know what things are called when I code. I just learn meself, haha. 
If you copy that handler function with the `(event, sender, data)` and subscribe it, just do something like
```
x = data.mvo.pos_x
y = data.mvo.pos_y
z = data.mvo.pos_z
```
inside the function.

lmmedino, 2020-05-07:
Hi, thanks for the help. I ended up getting the data, but I move Tello and it seems like the data is not changing, is this normal?

gemSquared, 2020-05-07:
If x,y, and z are near zero, hold your tello up about 0.5 meters over a well lit surface. Those values come up when the Tello is flying. They also may not start at zero because the values drift without mvo correction, so set the "home position" when in the air.

ALSO! When the tello loses sight of the ground (low light/flat texture) the mvo goes off and the values snap to zero again. Use the first byte of EVENT_LIGHT to get a low-light warning. If the byte equals 1, that means the mvo is about to/already cut off. Otherwise it'll be zero.

P.S. you may want to get yaw from the imu quaternions ;)

lmmedino, 2020-09-01:
Hi @gemSquared, 

Do you know where do we set up the frequency we get the data from tello?

jasonxytan, 2021-02-03:
Thank you! This error confused me for several days. @gemSquared 
Could you please update the source of pip so that people can pip install the latest version of tellopy package? @hanyazou

gemSquared, 2021-03-28:
> Hi @gemSquared,
> 
> Do you know where do we set up the frequency we get the data from tello?

It appears the tello just streams it in, hence why this library uses event handlers. Not 100% sure though.

closed 2022-05-24 by GalBrandwine