hanyazou/TelloPy #75: Error in video_effect.py

bhuvaneswarignanasekar, 2020-02-04:
Your code is great and most things work great but I tried video_effect.py and ran into some issues. It says log data is corrupted and then a timeout error. Would you help me to resolve it?. 

Tello: 12:49:30.588:  Info: start video thread
Tello: 12:49:30.588:  Info: send connection request (cmd="conn_req:9617")
Tello: 12:49:30.588:  Info: video receive buffer size = 425984
Tello: 12:49:30.588:  Info: state transit State::disconnected -> State::connecting
Tello: 12:49:30.635:  Info: connected. (port=9617)
Tello: 12:49:30.635:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 12:49:30.635:  Info: state transit State::connecting -> State::connected
Tello: 12:49:30.636:  Info: get video stream
Tello: 12:49:30.637:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:31.718:  Info: recv: log_header: id=2120, 'b'BUILD May  7 2019 12:02:11''
Tello: 12:49:31.819:  Info: LogData: UNHANDLED LOG DATA: id= 2064, length=  64
Tello: 12:49:31.819:  Info: LogData: UNHANDLED LOG DATA: id= 2208, length=  40
Tello: 12:49:31.819:  Info: LogData: UNHANDLED LOG DATA: id= 1000, length=  44
Tello: 12:49:31.819:  Info: LogData: UNHANDLED LOG DATA: id= 1001, length=  16
Tello: 12:49:31.820:  Info: LogData: UNHANDLED LOG DATA: id=10086, length=   4
Tello: 12:49:31.820:  Info: LogData: UNHANDLED LOG DATA: id=10085, length=  80
Tello: 12:49:31.820: Error: LogData: corrupted data at pos=1028, data=
Tello: 12:49:31.825: Error: LogData: corrupted data at pos=0, data=f5 f5 f5 f5 f5 f5 f5 f5 01 f4 7d e6 f5 f5 f5 f5 f5 f5 f5 f5 e4 f0 ef e4 f5 f5 1f 08 55 1c 00 7a e9 03 05 7f 86 00 05 05 05 05 05 05 05 05 05 05 05 05 05 05 d9 00 59 45 55 21 00 de 74 27 1b 7f 86 00 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 1b 10 db 55 84 00 cf 00 08 e6 f7 86 00 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 ed 20 48 a7 02 f4 bd da 82 eb 44 da c6 dc 99 59 3b ec 47 da 04 ae dd 5a f0 ad 62 5d e6 e6 e6 66 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 e6 11 f0 5a e4 af f3 e6 e6 e6 e6 e6 e6 79 e6 3f 54 55 4c 00 0d 10 08 37 f8 86 00 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 b7 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 37 3f 37 37 35 67 37 37 37 37 37 34 35 55 34 00 cd a0 08 6b f8 86 00 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b eb 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 6b 3b 6b 6b 6b 70 2a 55 10 00 37 66 27 67 26 87 00 67 67 67 67 cd 30 55 5c 00 e1 65 27 b4 26 87 00 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 b4 86 79 55 84 00 cf 00 08 9c 8b 8a 00 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 95 be 33 dd 8c a8 e7 a0 c4 1c 2e a0 c1 34 e3 23 d8 23 39 a0 c4 66 d0 20 6b ff 06 24 9c 9c 9c 1c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 9c 6a 8a 20 9e da 89 9c 9c 9c 9c 9c 9c 5b 9c 68 74 55 4c 00 0d 10 08 ce 8b 8a 00 ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce 4e ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce ce c6 ce ce c6 aa ce ce ce ce ce b6 d8 55 34 00 cd a0 08 02 8c 8a 00 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 82 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 02 66 02 02 02 94 e2 55 43 00 15 0c 00 b8 ef 8c 00 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 b8 58 a8 b8 a8 b8 b8 b8 bc b8 b8 b8 b8 b8 ac 32 a4 9f b8 b8 b8 b8 b8 b8 8f 35 55 38 00 80 e8 03 03 a6 8d 00 1d 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 f7 02 8b 10 03 03 03 03 03 03 03 03 12 06 19 12 03 03 83 d8 55 1c 00 7a e9 03 15 a6 8d 00 15 15 15 15 15 15 15 15 15 15 15 15 15 15 c9 10 c4 6e 55 84 00 cf 00 08 19 1f 8e 00 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 5c 2b b6 58 27 6d 46 25 5d da b4 25 f8 10 99 a6 63 00 b5 25 e9 07 5d a5 7b 5f 94 a3 19 19 19 99 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 eb 0f a5 1b 5f 0c 19 19 19 19 19 19 f6 19 34 d5 b9 f6
Tello: 12:49:31.826:  Info: LogData: UNHANDLED LOG DATA: id=   16, length=   4
Tello: 12:49:32.104:  Info: LogData: UNHANDLED LOG DATA: id= 1710, length=   6
Tello: 12:49:32.179:  Info: LogData: UNHANDLED LOG DATA: id=10100, length=  21
Tello: 12:49:32.281:  Info: LogData: UNHANDLED LOG DATA: id=   12, length=  55
Tello: 12:49:32.591: Error: video recv: timeout
Tello: 12:49:32.592:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:33.286:  Info: LogData: UNHANDLED LOG DATA: id= 1002, length=  27
Tello: 12:49:33.594: Error: video recv: timeout
Tello: 12:49:33.595:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:34.596: Error: video recv: timeout
Tello: 12:49:34.596:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:35.598: Error: video recv: timeout
Tello: 12:49:35.598:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:36.600: Error: video recv: timeout
Tello: 12:49:36.600:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:37.601: Error: video recv: timeout
Tello: 12:49:37.601:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:38.603: Error: video recv: timeout
Tello: 12:49:38.603:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:39.604: Error: video recv: timeout
Tello: 12:49:39.605:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:40.606: Error: video recv: timeout
Tello: 12:49:40.606:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:41.608: Error: video recv: timeout
Tello: 12:49:41.608:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:42.609: Error: video recv: timeout
Tello: 12:49:42.609:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:43.612: Error: video recv: timeout
Tello: 12:49:43.612:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:44.614: Error: video recv: timeout
Tello: 12:49:44.614:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:45.616: Error: video recv: timeout
Tello: 12:49:45.616:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:46.617: Error: video recv: timeout
Tello: 12:49:46.617:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:47.618: Error: video recv: timeout
Tello: 12:49:47.618:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:48.620: Error: video recv: timeout
Tello: 12:49:48.620:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:49.621: Error: video recv: timeout
Tello: 12:49:49.621:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:50.622: Error: video recv: timeout
Tello: 12:49:50.622:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:51.624: Error: video recv: timeout
Tello: 12:49:51.624:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 12:49:52.625: Error: video recv: timeout

ricafort, 2020-07-06:
@hanyazou @bhuvaneswarignanasekar, I get exactly the same issue and errors

samlaf, 2021-02-16:
Experiencing the same issue on Ubuntu 20.04 with python3.

Walt-H, 2021-02-16:
Check out my pull request here: https://github.com/hanyazou/TelloPy/pull/90

There's a bit you guys have to change to get it working.

I did my modifications on the `keyboard_and_video.py` example instead.

Walt-H, 2021-02-16:
To be more specific, check out the `README` from the fork here:
https://github.com/hanyazou/TelloPy/blob/112fec56854a3ac3ede813b3d99de20e4e0bc95f/README.md

Follow the instructions in the [keyboard_and_video section](https://github.com/hanyazou/TelloPy/blob/112fec56854a3ac3ede813b3d99de20e4e0bc95f/README.md#keyboard_and_video), and use the modified [keyboard_and_video.py](https://github.com/hanyazou/TelloPy/blob/112fec56854a3ac3ede813b3d99de20e4e0bc95f/tellopy/examples/keyboard_and_video.py).

samlaf, 2021-02-23:
I think I found a solution for this.
You just need to add a call to `wait_for_connection`:
```
drone.connect()
drone.wait_for_connection(10.0)
drone.start_video()
```
Alternatively, you can add the call to `wait_for_connection` directly inside `connect`, as I did in this [pull request](https://github.com/hanyazou/TelloPy/pull/96).

Helge543, 2021-08-19:
> @hanyazou @bhuvaneswarignanasekar, I get exactly the same issue and errors

I have exactly the same output