hanyazou/TelloPy #77: Error: video recv: timeout #29

CrashxZ, 2020-02-18:
Hi, I am trying to get the video from tello, and I am still getting this error ( Error: video recv: timeout)
Its the same from issue #29 

Here is the log:

mainak@mainak:~$ python -m tellopy.examples.video_effect
Tello: 02:13:07.741: Info: send connection request (cmd="conn_req:9617")
Tello: 02:13:07.742: Info: state transit State::disconnected -> State::connecting
Tello: 02:13:07.742: Info: start video thread
Tello: 02:13:07.742: Info: video receive buffer size = 425984
Tello: 02:13:07.771: Info: connected. (port=9617)
Tello: 02:13:07.771: Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 02:13:07.772: Info: state transit State::connecting -> State::connected
Tello: 02:13:07.775: Info: get video stream
Tello: 02:13:07.775: Info: start video (cmd=0x25 seq=0x01e4)
Tello: 02:13:07.821: Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 02:13:07.821: Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 02:13:07.822: Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 02:13:07.823: Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 02:13:13.745: Error: video recv: timeout
Tello: 02:13:18.750: Error: video recv: timeout
Tello: 02:13:23.756: Error: video recv: timeout
Tello: 02:13:28.769: Error: video recv: timeout
Tello: 02:13:33.769: Error: video recv: timeout
Traceback (most recent call last):
File "/home/mainak/.local/lib/python2.7/site-packages/tellopy/examples/video_effect.py", line 17, in main
container = av.open(drone.get_video_stream())
File "av/container/core.pyx", line 274, in av.container.core.open
File "av/container/core.pyx", line 226, in av.container.core.Container.cinit
File "av/container/core.pyx", line 123, in av.container.core.ContainerProxy.init
File "av/container/core.pyx", line 194, in av.container.core.ContainerProxy.err_check
File "av/utils.pyx", line 105, in av.utils.err_check
AVError: [Errno 1094995529] Invalid data found when processing input: ''
[Errno 1094995529] Invalid data found when processing input: ''
Tello: 02:13:37.790: Info: quit
Tello: 02:13:37.790: Info: state transit State::connected -> State::quit
<class 'tellopy._internal.video_stream.VideoStream'>.handle_event(DISCONNECTED)
Tello: 02:13:37.843: Info: exit from the recv thread.
Tello: 02:13:38.771: Error: video recv: timeout
Tello: 02:13:38.771: Info: exit from the video thread.

ricafort, 2020-07-06:
@hanyazou . I am having same issue too

tufeixp, 2020-09-28:
@hanyazou , same here on windows 10 @ python 3.8

Tello: 23:09:15.264:  Info: start video thread
Tello: 23:09:15.264:  Info: send connection request (cmd="conn_req:9617")
Tello: 23:09:15.298:  Info: video receive buffer size = 524288
Tello: 23:09:15.303:  Info: state transit State::disconnected -> State::connecting
Tello: 23:09:15.323:  Info: connected. (port=9617)
Tello: 23:09:15.333:  Info: send_time (cmd=0x46 seq=0x01e4)
Tello: 23:09:15.342:  Info: state transit State::connecting -> State::connected
Tello: 23:09:15.343:  Info: get video stream
Tello: 23:09:15.353:  Info: start video (cmd=0x25 seq=0x01e4)
Tello: 23:09:15.375:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 23:09:15.387:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 23:09:15.399:  Info: recv: ack: cmd=0x34 seq=0x0000 cc 60 00 27 90 34 00 00 00 00 72 a5
Tello: 23:09:15.405:  Info: recv: ack: cmd=0x20 seq=0x0000 cc 60 00 27 b0 20 00 00 00 00 42 b9
Tello: 23:09:31.330: Error: video recv: timeout
Tello: 23:09:46.433: Error: video recv: timeout
Tello: 23:10:01.441: Error: video recv: timeout
Tello: 23:10:16.460: Error: video recv: timeout
Tello: 23:10:31.500: Error: video recv: timeout

Walt-H, 2020-09-29:
I had the same issues, comments here worked for a few of us Windows users: https://github.com/hanyazou/TelloPy/issues/78#issuecomment-660214525

adrienJeg, 2020-10-06:
Same issue on Ubuntu 18.04 and Python 3.8

AlessandroRuggiero, 2020-11-05:
I had the same issue for months, no solutions so far....

Walt-H, 2020-11-16:
#90

samlaf, 2021-02-23:
I think I found a solution for this.
You just need to add a call to `wait_for_connection`:
```
drone.connect()
drone.wait_for_connection(10.0)
drone.start_video()
```
Alternatively, you can add the call to `wait_for_connection` directly inside `connect`, as I did in this [pull request](https://github.com/hanyazou/TelloPy/pull/96).

Helge543, 2021-08-19:
@samlaf are you sure that your solution works? I also tried this in the keyboard_and_video.py example but it did not work. What do you mean by "Alternatively, you can add the call to wait_for_connection directly inside connect"? you mean like this ... `drone.connect(drone.wait_for_connection(10.0))` ... because this does not work.

samlaf, 2021-08-20:
Yes that is what I meant, as you can see from looking at the pull request.
I dont quite remember the details though... its been a while, sorry.

closed 2024-07-18 by CrashxZ