hanyazou/TelloPy #78: chaning the directory for pictures and video

trooperisme, 2020-03-19:
Hi, This is a very very easy easy question, but I dont know how to work with '%s'. Let's say that my directory is: C:\Users\hi\Pictures\fly

trooperisme, 2020-03-19:
> path = '%sUsers/hi/Pictures/Screenshots/tello-%s.jpeg' % (
        'C:/',
        datetime.datetime.now().strftime('%Y-%m-%d_%H%M%S'))
    with open(path, 'wb') as fd:
        fd.write(data)
    status_print('Saved photo to %s' % path)  ### I
## I switch the os.getenv() method back to 'C:/'. Then I end up getting the message:

> Tello: 13:25:14.920: Info: unknown packet: 0030 cc 60 00 27 b0 30 00 00 00 00 02 0d
Tello: 13:25:14.921: Info: recv: file size: cc 90 00 be 88 62 00 2e 00 01 d8 d5 09 00 00 00 2b 5f
Tello: 13:25:14.921: Info: file size: num=0 bytes=644568
ALT: 10 | SPD: 0 | BAT: 61 | WIFI: 90 | CAM: 0 | MODE: 6
Tello: 13:25:14.971: Info: recv: file size: cc 60 00 27 50 62 00 2e 00 00 db ff
Tello: 13:25:14.972: Warn: file size: payload too small: 00.

##Then the camera screen starts to freezing. I am so confused now, also I cannot record video either (mostly due to the 'path')
Thanks,

viet2411, 2020-05-22:
Hello,
I also changed the os.getenv() method back to 'C:/', but I received the file not found error.
Have you solved this problem yet?

trooperisme, 2020-05-22:
Hi Viet, I tried but it still doesn’t work out. I have been playing around with the path but I cant change the path. I think you should keep it as default

viet2411, 2020-05-22:
Update:
I switched the os.getenv() method to os.getcwd() and was able to capture a picture, but the camera screen freezed after that. Still don't understand why this happenned lol

trooperisme, 2020-05-22:
> Update:
I switched the os.getenv() method to os.getcwd() and was able to capture a picture, but the camera screen freezed after that. Still don't understand why this happenned lol

that is exactly what I experienced (you still can have access to control via your device, it is just the camera doesn't get the images)

trooperisme, 2020-05-22:
@hanyazou please help if you feel free

gemSquared, 2020-05-22:
Hello! I just removed all the previous directory jazz and did `"telloPic" + str(round(time.time())) + ".png"` for the path. It'll save to the same folder as the python script.

trooperisme, 2020-05-22:
oh thats nice, it makes more sense to me tho

viet2411, 2020-07-07:
Has anyone recorded the video yet?

Walt-H, 2020-07-17:
@viet2411 I was able to do this through Windows :)

I had to do the following to setup my environment:

1.  Download Mplayer for Windows here: [https://sourceforge.net/projects/mplayerwin/files/MPlayer-MEncoder/r38151/mplayer-svn-38151-x86_64.7z/download](https://sourceforge.net/projects/mplayerwin/files/MPlayer-MEncoder/r38151/mplayer-svn-38151-x86_64.7z/download)

2. Unzip the file via [7zip](https://www.7-zip.org/)

3. Add the extracted folder to my PATH (I followed [this guide](https://www.architectryan.com/2018/03/17/add-to-the-path-on-windows-10/))

4.  Installed the wheel (see README) and the following packages into my Python environment via pip: **av**, **opencv-python**, **image**, **pygame**

5. Changed **filename** to a static directory I wanted (i.e. `filename = '%s\\Pictures\\tello-%s.mp4' % ('C:\\Users\\Walt
[image]
', datetime.datetime.now().strftime(date_fmt))` )

6.  In [keyboard_and_video.py](https://github.com/hanyazou/TelloPy/blob/develop-0.7.0/tellopy/examples/keyboard_and_video.py), I uncommented line **62 and 63**

7. Run [keyboard_and_video.py](https://github.com/hanyazou/TelloPy/blob/develop-0.7.0/tellopy/examples/keyboard_and_video.py)

8. Hit **R**

9. Fly Around a Little

10. Hit **R**

And now I have an MP4 that I have to open through Google Chrome to view :)

viet2411, 2020-07-18:
@Walt-H Finally I can record video with this program. Thank you so much for your support!

Walt-H, 2020-11-01:
#90 was made to help address