hanyazou/TelloPy #80: Is it possible to get X,Y,Z position from tello using tellopy?

lmmedino, 2020-04-12:


gemSquared, 2020-04-21:
https://github.com/hanyazou/TelloPy/issues/72#issuecomment-617052742

Comment back here if you need more help getting this data! 😄

Walt-H, 2020-10-23:
The Tello doesn't have a GPS, you can't really know your point in space, but there is a provided estimate from your initial hover point (position after lift off).

[#72 (comment)](https://github.com/hanyazou/TelloPy/issues/72#issuecomment-617052742) explains everything well, but I did the following:

Inside the return value of **drone.EVENT_FLIGHT_DATA**, you can get positional and velocity data via an event handler:
```
def handler(event, sender, data, **args):
    drone = sender
    if event is drone.EVENT_FLIGHT_DATA:
        # perform actions here
        print(data)

# some code later....

drone.subscribe(drone.EVENT_FLIGHT_DATA, handler)
```

dullahgtt, 2022-06-16:
If I created a function that defined the best possible path between two points and I wanted to translate that path into directions for the tello drone to fly, how would I do that. I do not have any positional data on the drone nor do I have any directional data, which makes it hard to figure out which direction the drone needs to go initially to follow the designated path.