hanyazou/TelloPy #87: fly multiple tello with video in AP mode

snakehaihai, 2020-09-21:
Hi 
Assume i have N usb wifi adapter and N tello. How do i use this package to run multiple of Tello concurrently? I tried 2 USB + 2 tello. when i launch the first one and enable the video feed. I cant get the second video feed.

Walt-H, 2020-09-21:
Assuming 2 USB Wifi devices and Tellos, I would connect each network device to a unique Tello drone.

Then I would modify how tello.py binds to network interface starting [here](https://github.com/hanyazou/TelloPy/blob/develop-0.7.0/tellopy/_internal/tello.py#L99).

I'd use the below function to get the IP of my network interface:
```
import socket 
import fcntl 
import struct 
 
def get_ip_address(ifname): 
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM) 
    return socket.inet_ntoa(fcntl.ioctl( 
        s.fileno(), 
        0x8915, 
        struct.pack('256s', ifname[:15]) 
    )[20:24]) 
```

Then change my binding as so:
```
s = socket.socket() 
s.bind((get_ip_address('eth0'), 8889))  # change to your ifconfig settings
```

Lastly, I'd run two separate instances of lets say the keyboard_and_video example, where the binding addresses (aka network interfaces) are different, along with the mp4 file names.

snakehaihai, 2020-09-21:
> Assuming 2 USB Wifi devices and Tellos, I would connect each network device to a unique Tello drone.
>
> Then I would modify how tello.py binds to network interface starting [...]

Thanks for the quick reply. 
Then for 2 tello + wifi.  How do i set the wifi adapter? just let it DHCP?

snakehaihai, 2020-09-21:
> Assuming 2 USB Wifi devices and Tellos, I would connect each network device to a unique Tello drone.
>
> Then I would modify how tello.py binds to network interface starting [...]

I think because of 2 tello is trying to connect to the same 192.168.10.1 address. that's why. one of the adapter is constantly lost connection. in this case. how do i set it up?

snakehaihai, 2020-09-21:
> Assuming 2 USB Wifi devices and Tellos, I would connect each network device to a unique Tello drone.
>
> Then I would modify how tello.py binds to network interface starting [...]

hey i just tried your suggestion. But turns out only 1 drone is in command mode/ AKA light turns green.  it means the code only works for 1 drone, no matter it is same python code or put it separately.  I think need to solve the adapter IP assignment and port forwarding issue first.

[image]

Walt-H, 2020-09-21:
Hi @snakehaihai ,

My solution only works on Windows (my current box).

On Linux the interface is normally not chosen until a routing decision is made, therefore you usually don't have a say on which interface your packets leave. You do have the option though of using SO_BINDTODEVICE (see **man 7 socket**) on Linux. This binds a socket to a device.

You can do the following to bind it to a specific device (need **root**):

```
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.setsockopt(socket.SOL_SOCKET, 25, 'eth0')
```

Hope this helps

snakehaihai, 2020-09-22:
> Hi @snakehaihai ,
> 
> My suggestion only works on Windows (my current box).
> 
> On Linux the interface is normally not chosen until a routing decision is made, therefore you usually don't have a say on which interface your packets leave. You do have the option though of using SO_BINDTODEVICE (see **man 7 socket**) on Linux. This binds a socket to a device.
> 
> You can do the following to bind it to a specific device (need **root**):
> 
> ```
> s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
> s.setsockopt(socket.SOL_SOCKET, 25, 'eth0')
> ```
> 
> **Note**: sockets are C bindings, so from `socket.h` the **SO_BINDTODEVICE** option is 25
> 
> Hope this helps

thank you all the same. Cause I`m flying multiple tello with each drone running an orbslam instance to achieve real-time collaborative image-based localization and mapping for automated path planning and decision. All of the package are only tested in ubuntu. therefore, I cant use windows.

Walt-H, 2020-09-22:
@snakehaihai 
Yes, that last comment will more than likely work on your Linux box (**SO_BINDTODEVICE** option), I just don't have a Linux box to test on.

Also, that's pretty cool, I saw some of the repos on your page and they look fun!

Walt-H, 2020-09-29:
@snakehaihai 
Just curious if this ever worked :D

snakehaihai, 2020-09-30:
> @snakehaihai
> Just curious if this ever worked :D

not exactly.
Im using a NUC for each tello drone.  and 3 NUC is connecting to the same Server PC that does the coordination.  The video in each pc is obtained from the same ip. say 192.168.10.1  for PC1 PC2 PC3. and this video is being port forward by FFMEPG into other IP say 192.168.9.1 for PC1 and 192.168.8.1 for PC2...  then it works. same goes for reverse tello control . It is slightly more complicated but it works.