hanyazou/TelloPy #88: Help with mplayer

H4CKL0RD, 2020-10-21:
What is mplayer and how do I install it?

H4CKL0RD, 2020-10-21:
Also because I am getting an error that says no file or directory found "mplayer"

Walt-H, 2020-10-21:
See my comment here: [changing the directory for pictures and video
](https://github.com/hanyazou/TelloPy/issues/78#issuecomment-660214525)

I am working on a fork to make this platform independent :)

H4CKL0RD, 2020-10-21:
Ok thanks!

H4CKL0RD, 2020-10-21:
The program you listed was for windows only, I am using a mac. Do you know any alternatives for mac?

H4CKL0RD, 2020-10-21:
Oh nevermind to use on mac you had to install it with homebrew
brew install mplayer

Walt-H, 2020-10-21:
I'll be sure to throw that in there ;)

Walt-H, 2020-10-21:
Did the brew install resolve your issues?

H4CKL0RD, 2020-10-21:
Not fully, I still have a black screen but I can control the drone fine. So I am not getting a video feed properly.

suakii, 2020-10-23:
I have the same issue on mac.
After start program I have two windows.

Walt-H, 2020-10-23:
It could be something going on at these two places:

- In keyboard_and_video.py, filename at [Line 57](https://github.com/hanyazou/TelloPy/blob/develop-0.7.0/tellopy/examples/keyboard_and_video.py#L57) uses an env variable that Python may not have initialized. I recommend making this path static as a quick fix
     - I would recommend doing this for jpegs at line [209](https://github.com/hanyazou/TelloPy/blob/develop-0.7.0/tellopy/examples/keyboard_and_video.py#L209)
- Maybe uncomment line [62 and 63](https://github.com/hanyazou/TelloPy/blob/develop-0.7.0/tellopy/examples/keyboard_and_video.py#L62) (may not be necessary)