hanyazou/TelloPy #92: Are the POS values thrust worthy?

GalBrandwine, 2020-11-14:
Hey @gemSquared, @hanyazou 
I'm currently working on a project that implements all  TelloPy's capabilities in *c++*,
I need these capabilities as part of my *autonomous tello using **ROS2*** project.
For this project, I use regular Tello ( Not EDU).

# Here's the thing: 
I noticed some weird behavior in this so-called **POS[x,y,z]** that the drone send's back.
Have you noticed any of it?

To test this I changed line 392 in `protocol.py` *quick & dirty changes :)*
```python
self.log.info('LogNewMvoFeedback: ' + str(self))
```
And I ran the following script **several times**:
```python
from time import sleep
import tellopy

def handler(event, sender, data, **args):
    drone = sender
    if event is drone.EVENT_LOG_DATA:
        pass

def test():
    drone = tellopy.Tello()
    try:
        drone.subscribe(drone.EVENT_LOG_DATA, handler)   
        drone.connect()
        drone.wait_for_connection(60.0)

        drone.takeoff()
        sleep(5)
        drone.down(50)
        sleep(5)
        drone.land()
    except Exception as ex:
        print(ex)
    finally:
        drone.quit()

if __name__ == '__main__':
    test()
```

## Expectations
On the ground it looks like: `0.02  0.01 -0.00` which is somewhat expectable.
First time running:
upon reaching  **takeoff height**: `1.91  0.95  1.19`
upon **down(50)**: `1.98  0.95  2.16`

The drone landed, now I ran the script again without moving the drone/restarting/etc...
upon reaching  **takeoff height**:  `4.93  7.50  9.98`
upon **down(50)**: `5.01  7.48 10.45`

Some question that  must be asked:
* Why these values aren't the same in both scenarios? meaning the drone did the exact same thing.
* Why the **Z** value is not accurate? - -Isn't it taken directly from the IR sensor?

## Conclusions
This drone obviously has some localization solution. And not a bad one. 
I assume that the **pos** data is the localization solution. Am I right? 
If so, what am I doing wrong to receive such bad **pos** readings?

Thanks, Gal

Walt-H, 2020-11-16:
You should definitely **not** trust it.

With the Tello being around $100 dollars, the error margins of its sensors are going to be pretty big, but they do a good job compensating through software, but of course it's **not** applicable to real applications.

Your best bet is to probably to use an object with known dimensions and with a static, 3D location, where you can detect it via OpenCV and localize the drone with some trigonometry.

This is a good article on [tracking via OpenCV](https://www.pyimagesearch.com/2015/05/04/target-acquired-finding-targets-in-drone-and-quadcopter-video-streams-using-python-and-opencv/).

Walt-H, 2020-11-16:
On the note of C++, you should check out these two repos:

- https://github.com/carlospzlz/ctello
- https://github.com/clydemcqueen/tello_ros

I've successfully used the first one, but I am planning on trying the ROS version soon.

GalBrandwine, 2020-11-16:
Hey @Walt-H, 
Thanks for the quick reply!

As for trusting the **pos** output or not - I think you're right. 
I guess some creativity needed here 🤓, the solution you have suggested will force me to add external objects though.
I aim to solve this one without external help (or else I could just use a MOCAP system, or another localization sensing, anyway - not the issue here)

I do know these *repo's* you've suggested above, for my knowledge they do not support the unofficial **SDK**.

Thanks for sharing your opinion!
Gal

Walt-H, 2020-11-17:
I see what you mean, you want to make a self contained package, which takes a lot of the pain out of new users configuring their computers.

The Tello does not support the unofficial, nor the SDK 2.0, but I did find the SDK 1.3 documentation for others to find [here](https://terra-1-g.djicdn.com/2d4dce68897a46b19fc717f3576b7c6a/Tello%20%E7%BC%96%E7%A8%8B%E7%9B%B8%E5%85%B3/For%20Tello/Tello%20SDK%20Documentation%20EN_1.3_1122.pdf).

Your ROS2 project looks really cool, and hope it turns out great!

Best,
Walt

GalBrandwine, 2020-11-18:
I think it's an interesting question, so keeping this issue open for now..

closed 2022-05-24 by GalBrandwine