hanyazou/TelloPy #95: Can't use this low-level protocol and official SDK at the same time :(

samlaf, 2021-02-23:
I'm trying to use the [mission pads](https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20Mission%20Pad%20User%20Guide.pdf) that come with the Tello EDU.

The Mission Pads requires using the official [SDK2.0 protocol](https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf).

The official sdk streams videos to udp port 11111, and state information to port 8890.
The low-level protocol from this repo streams videos to whatever channel it receives in the initial "conn_req:PORT" command, and streams state information (through the command 0x0056 - Flight Status) to the port it receives commands from (9000 in this repo).

However, as soon as the Tello receives a "command" command to initiate the official SDK, it stops streaming on the low-level protocol ports, and furthermore refuses to acknowledge future low-level "conn_req:PORT" commands. Essentially, as soon as the official sdk takes over, we stop receiving video and state streaming from the low-level protocol.

Has anyone found a solution for this?

samlaf, 2021-02-23:
Seems like this was already raised in #50.

Walt-H, 2021-02-23:
I was able to do what you described using the this [damiafuentes/DJITelloPy](https://github.com/damiafuentes/DJITelloPy), please see this class:

https://github.com/damiafuentes/DJITelloPy/blob/9c3f61bc6b4f0c79c01c3599396d921fd8276dc4/djitellopy/tello.py#L21

It defines the ports a little different that you:
```
VS_UDP_PORT = 11111
CONTROL_UDP_PORT = 8889
STATE_UDP_PORT = 8890
```

I was then able to contribute this mission pad PoC file for that project:
https://github.com/damiafuentes/DJITelloPy/blob/master/examples/mission-pads.py

Maybe you can modify your fork of hanyazou/TelloPy project to do something like this.

Walt-H, 2021-02-23:
Also, no mission pad code has been added to this repo. If you end doing something that works with this library, maybe you can send a PR :D

It may not get merged anytime soon, but people like me would find it and definitely appreciate it!

samlaf, 2021-02-23:
Hey @Walt-H !
I've also been using the DJITelloPy repo for mission pads up to now, but I can't stand the 2s latency on the video stream of the sdk protocol (still hoping to find a way to bring it down... if I did I wouldn't need this repo at all).

I think I'll try to use apriltags for now instead of the mission pads. If I can't get it to work properly I might then as a last resort try to reverse engineer the low-level mission pad protocol.

Walt-H, 2021-02-23:
@samlaf I misconstrued the two ports lol

Walt-H, 2021-02-23:
Apparently the Tello EDU phone app now uses mission pads, this app also uses 6038 for video, so I am going to do some JAR decompiling and try to find some details for you buddy :D

Walt-H, 2021-02-23:
Apparently the Tello EDU phone app now uses mission pads, this app also uses 6038 for video, so I am going to do some JAR decompiling and try to find some details for you buddy :D

Could you by chance run the Tello EDU app on your phone or simulator and subsequently sniff that traffic?

samlaf, 2021-02-23:
Awesome stuff!
I can do that. I never really figured out how to use that app though, what exactly do I need to do and what packets do you want me to sniff? (video only or mission pads also? etc)

Also, JAR decompiling?? You're running it on an android and think you can find the relevant information from decompiling the code, without sniffing the protocol?

Walt-H, 2021-02-23:
I meant to say do you know how to sniff traffic on your phone; however, I've read if you run an emulator, you can just sniff the traffic on your computer on the addresses/ports of interest.

Since there are lot of Java decompilers, I just download the app on my emulator, compile it, and looking then decompiling the JAR. Just wondering if I could find any code that focuses on those ports, or have signatures that can be seen in data from sniffed packets.

I use Android Studio emulators, since I don't have a Mac to run iPhone ones.

samlaf, 2021-02-23:
Ah! That sounds like a big headache... :o
Did you know that you can change your network adapter card to run in monitor mode and sniff traffic between the tello and phone? That's what I was doing last week.

Walt-H, 2021-02-25:
After playing with it and looking at different traffic, I observed the  same thing in Tello EDU: when I want to go use the mission pads, all low-level streams stop, and camera functions in the app no longer work (take photo/record).

I guess it's devoting all of its tiny computational power to image processing/recognition. I see that there is also additional data on the comms port, but sounds like a lot of work to decipher. 

IMO I think the Tello isn't the drone for your requirements. To keep costs low, you should use [ArduPilot](https://ardupilot.org/) and follow tutorials to build your own small, inexpensive drones. There are a lot of drone frames available to freely 3D print, along with propellers, movable Arduino camera mounts, etc.

Once you build one, just repeat the process until you have a swarm; in addition, you control EVERYTHING :D

juanma-t3j, 2022-08-30:
thanks mannnn, yours replys have helped me